GPUOpen-Drivers/AMDVLK #84: 4 games tested

enthalpie, 2019-02-24:
1, Rise of the Tomb Raider dies with a sigabort 6, error. 
The other 3 are here: https://www.youtube.com/watch?v=D4SY2spx5g0

Fallout76 wouldnt start with the AMDVLK config file for the overlay present, just crashed right after launch.

Mad Max works great ! All in all the fps feels pretty native like on win. Yet its russian roulette if a game works or not. 

FYI, keep up the good work !

jinjianrong, 2019-02-26:
@enthalpie Rise of the Tomb Raider: is this the DX version running on dxvk or the native Vulkan version?

enthalpie, 2019-03-13:
Native Vulkan and I was able to choose the AMDVLK driver from the Feral launch menu upon startup
I compared for more games in direct comparison with RADV: https://www.youtube.com/watch?v=gpG0B977VzY

jinjianrong, 2019-03-13:
@enthalpie we had a fix for Rise of the Tomb Raider crash, could you help check if the latest driver works well with the game?

Johnnynator, 2019-03-13:
The issue is still existent for me (amdvlk-2019.Q1.7, RX 580)

jinjianrong, 2019-03-14:
@Johnnynator are both AMDVLK and RADV driver installed on your system?

Johnnynator, 2019-03-14:
> @Johnnynator are both AMDVLK and RADV driver installed on your system?

Yes both are.

JacobHeAMD, 2019-03-14:
Hi @Johnnynator , could you please tell me how to choose AMDVLK or RADV in the menu of "Rise of the Tomb Raider"? I didn't find any option in menu to setting vendor driver.

Johnnynator, 2019-03-14:
Idk, I just went setting `VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/amd_icd64.json %command%` in the Games Launch Options inside Steam.

Johnnynator, 2019-03-15:
ROTTR runs fine now with 2019.Q1.8 (gcc 8.3.0).

JacobHeAMD, 2019-03-18:
Thanks for your updating!

enthalpie, 2019-03-18:
@jinjianrong Amazing the new driver runs very well ! Rise of the Tomb Raider is working now, I ran the benchmark and it got 105 fps on AMDVLK and 95 fps on RADV, so its outperforming. 
I tried Star Citizen too and that is working now (died with an 'incompatible GPU error' before). Lots of shader caching in the first 10 minutes and I set the shader to disc enabled in amdPAL.cfg but after that its working, even more solid than on RADV is my impression. Runs with a constant 35 - 60 fps what is good for that game. 
Devil May Cry 5 is working at 60 fps constant. 
Mad Max seems to be improved, runs most of the times around 75 fps. 
Pretty solid ! Only Fallout 76 has graphical errors still, tho lesser than from what I have tested. 
I will try more games, when I find the time. Keep up the good work :).

One question: I tried to test Anthem and Battlefield V but the origin launcher ends in a GPU hang. When I launch it the loading windows appears, then the launcher openes up, starts loading its content, flickers in black and white, turns black and the GPU stalls. I tried with overlays on and all off, same result.

kattjevfel, 2020-07-11:
For what it's wroth I can't get Fallout 76 running with amdvlk still, right as I get in-game my xorg session locks up and I have to kill it.
```
[ 2390.574013] [drm:amdgpu_dm_atomic_commit_tail [amdgpu]] *ERROR* Waiting for fences timed out!
[ 2391.086619] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, but soft recovered
[ 2401.326201] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, but soft recovered
[ 2411.566214] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, but soft recovered
[ 2421.806449] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, but soft recovered
[ 2432.046548] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, but soft recovered
[ 2442.286715] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, but soft recovered
[ 2452.526639] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, but soft recovered
[ 2462.766713] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, but soft recovered
[ 2504.239205] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, but soft recovered
```

Launch options: `VK_ICD_FILENAMES=/usr/share/vulkan/icd.d/amd_icd32.json:/usr/share/vulkan/icd.d/amd_icd64.json DXVK_HUD=full WINEDLLOVERRIDES="winedbg.exe=d" %command%`

jinjianrong, 2020-08-04:
The Fallout 76 issue is fixed by https://github.com/GPUOpen-Drivers/llpc/commit/b92193cbbe7e33521e1e720b68a3b53ae43d0bc7

closed 2020-08-11 by ntropy83

kattjevfel, 2020-08-19:
Sadly that does not appear to be the case, I just tried running Fallout 76 with amdvlk 2020.Q3.3 and it locked up my whole computer just as I got in-game.
All I was able to get out of dmesg:
```
[ 5304.255192] [drm:amdgpu_dm_atomic_commit_tail [amdgpu]] *ERROR* Waiting for fences timed out!
[ 5306.303301] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, but soft recovered
[ 5316.543088] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, but soft recovered
```

Flakebi, 2020-08-19:
I think this commit is not yet on master, so it’s not included in the 2020.Q3.3 release.
It should be in the next release or the one after that (I’m not quite sure until when commits get included).

If you want to test it, you have to build the dev version.

kattjevfel, 2020-08-25:
2020.Q3.4 mentions 
> Fallout 76 Game : System hang observed at Loading screen

with that installed the only "improvement" I can see is that I instead get pretty colours all over my screen, and I was able to recover by killing Xorg and re-logging.
https://gist.github.com/kattjevfel/e86948803967d7d957f7996c1bca99e4

jinjianrong, 2020-09-14:
@kattjevfel could run Fallout 76 with 2020.Q3.5 to verify if the issue is fixed?

kattjevfel, 2020-09-16:
@jinjianrong Works great with 2020.Q3.5! Only some slight visual glitches here and there, but from what I've seen those are old issues also appearing in amdgpu pro.