GPUOpen-Drivers/AMDVLK #86: Would you make packages for Fedora..

lejeczek, 2019-02-26:
..for us lazy ones, that would be fantastic.
many thanks, L.

tomkv, 2019-04-06:
FYI, there's a package in Fedora copr.

cmdrogogov, 2019-06-06:
> FYI, there's a package in Fedora copr.

You probably already know this, but It appears the Fedora30 packages failed the build process on this copr repo;

https://copr.fedorainfracloud.org/coprs/tkov/amdvlk/package/amdvlk-vulkan-driver/

tomkv, 2019-06-06:
Yes, it is failing due to #89, as F30 has gcc9 as it's compiler. The build currently running will also fail. You could try the F29 version (I installed it in F29 and then updated the distro; it still works).

samuelig, 2020-04-10:
Is there any plan to have official .rpm packages of AMDVLK?

JaxLinAMD, 2020-07-24:
rpm packages is provided since v-2020.Q3.1 @samuelig

samuelig, 2020-07-24:
Great, I can confirm it works on a Fedora system. Would it be possible to provide a RPM repo like Ubuntu's? It would be a great idea to avoid installing manually the .rpm packages.

JaxLinAMD, 2020-07-24:
Sorry, I don't have the plan to set up a server and deploy RPM repo for now :)

closed 2020-07-24 by JaxLinAMD