GPUOpen-Drivers/AMDVLK #88: Missing H.264 and VC-1 capabilities on Tahiti chipset

mflayer, 2019-03-13:
The issue: when using AMDVLK instead of RADV, H.264 and VC-1 hardware decoding/encoding capabilities are not recognized.
System: Arch linux (5.0 kernel), Mesa 18.3, latest available AMDVLK build (r62...)
Chipset in question: Radeon R9 280

Notes: I had to use `radeon.si_support=0 amdgpu.si_support=1` to force-load amdvlk. Don't know how that affects the decoding support.

Attached are the vainfo and vdpauinfo outputs for both drivers.

[vainfo_amdvlk.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/2962502/vainfo_amdvlk.txt)
[vainfo_radeon.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/2962503/vainfo_radeon.txt)
[vdpauinfo_amdvlk.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/2962504/vdpauinfo_amdvlk.txt)
[vdpauinfo_radeon.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/2962505/vdpauinfo_radeon.txt)

mflayer, 2019-03-14:
I realise I might have made a mistake. This issue lies with the amdgpu rather than with the amdvlk driver, doesn't it? I apologize, if it does. Hours of googling made my brain soft.

JacobHeAMD, 2019-03-14:
Hi @mflayer , Vulkan doesn't support video encoding/decoding. vainfo should report the video capability according to the video driver, such as radeonsi_drv_video.so.

closed 2019-03-14 by JacobHeAMD