GPUOpen-Drivers/AMDVLK #89: build failture on fedora 30 with gcc 9.0.1

andrew-ld, 2019-05-07:
error: https://privatebin.net/?bda738ee50d14e4f#FaVmjUHHVEirh9gSVzmMiUuLGW30TW9jew5bE4M8OQU=

> gcc -v
```
Using built-in specs.
COLLECT_GCC=/usr/bin/gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/9/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,objc,obj-c++,ada,go,d,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl --enable-offload-targets=nvptx-none --without-cuda-driver --enable-gnu-indirect-function --enable-cet --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux
Thread model: posix
gcc version 9.0.1 20190312 (Red Hat 9.0.1-0.10) (GCC) 
```

ChristophHaag, 2019-06-24:
Indeed, please don't use -WError in a release build when many people will build it with more recent compilers. Perhaps you can add an extra "no warning" mode that can be enabled by CI or so.

amdvlk builds for me on arch with
```bash
for i in xgl/icd/CMakeLists.txt llpc/CMakeLists.txt llpc/imported/metrohash/CMakeLists.txt llvm/utils/benchmark/CMakeLists.txt llvm/utils/benchmark/test/CMakeLists.txt pal/src/core/imported/addrlib/CMakeLists.txt pal/src/core/imported/vam/CMakeLists.txt pal/shared/gpuopen/cmake/AMD.cmake
do
  sed -i "s/-Werror//g" "$srcdir"/$i
done
```

JaxLinAMD, 2020-07-24:
close it hence it's an old ticket.

closed 2020-07-24 by JaxLinAMD