GPUOpen-Drivers/AMDVLK #90: Add step to pre-built driver installation instructions

viggy96, 2019-05-27:
Might want to mention uninstalling the 'mesa-vulkan-driver' package in the instructions. Because if you don't, you'll still be using Mesa, and not AMDVLK. 

I can confirm this, with Ubuntu 18.04.

Johnnynator, 2019-05-27:
You can easily change the uses driver by setting the `VK_ICD_FILENAMES ` environment var to the driver you want to use.
https://github.com/KhronosGroup/Vulkan-Loader/blob/master/loader/LoaderAndLayerInterface.md#table-of-debug-environment-variables

viggy96, 2019-05-27:
What would I set it to if I used the deb package? The instructions are clear for if you built the driver yourself, but its unclear what the variable value should be if I used the pre-built release of AMDVLK.

Johnnynator, 2019-05-27:
Most likely `/usr/share/vulkan/icd.d/amd_icd64.json`, but you should verify it before, just list all the files that are installed with the deb package and whatever `.json` file gets listed is it.

closed 2020-08-19 by JacobHeAMD