GPUOpen-Drivers/AMDVLK #91: Add Support for Vega 8 in Mobile Raven Ridge

sp82, 2019-05-28:
I have a notebook with a Ryzen 2500U with Vega 8 and vulkaninfo say Unsupported GPU.

sp82, 2019-05-29:
This is what a get from dmesg:

[25291.093478] gmc_v9_0_process_interrupt: 27 callbacks suppressed
[25291.093487] amdgpu 0000:02:00.0: [gfxhub] no-retry page fault (src_id:0 ring:158 vmid:2 pasid:32774, for process vulkan-smoketes pid 28192 thread vulkan-smoketes pid 28192)
[25291.093496] amdgpu 0000:02:00.0:   in page starting at address 0x0000e497c0329000 from 27
[25291.093499] amdgpu 0000:02:00.0: VM_L2_PROTECTION_FAULT_STATUS:0x0020153C
[25296.560553] [drm:amdgpu_dm_commit_planes.isra.43 [amdgpu]] *ERROR* Waiting for fences timed out.
[25301.178540] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, signaled seq=1691807, emitted seq=1691809
[25301.178623] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* Process information: process vulkan-smoketes pid 28192 thread vulkan-smoketes pid 28192
[25301.178626] [drm] GPU recovery disabled.
[25301.680578] [drm:amdgpu_dm_commit_planes.isra.43 [amdgpu]] *ERROR* Waiting for fences timed out.
[25306.800591] [drm:amdgpu_dm_commit_planes.isra.43 [amdgpu]] *ERROR* Waiting for fences timed out.
[25311.418777] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, signaled seq=1691807, emitted seq=1691809
[25311.418865] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* Process information: process vulkan-smoketes pid 28192 thread vulkan-smoketes pid 28192
[25311.418868] [drm] GPU recovery disabled.
[25321.658760] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, signaled seq=1691807, emitted seq=1691809
[25321.658846] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* Process information: process vulkan-smoketes pid 28192 thread vulkan-smoketes pid 28192
[25321.658850] [drm] GPU recovery disabled.
[25331.898487] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, signaled seq=1691807, emitted seq=1691809
[25331.898574] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* Process information: process vulkan-smoketes pid 28192 thread vulkan-smoketes pid 28192
[25331.898577] [drm] GPU recovery disabled.
[25342.138439] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, signaled seq=1691807, emitted seq=1691809
[25342.138526] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* Process information: process vulkan-smoketes pid 28192 thread vulkan-smoketes pid 28192
[25342.138529] [drm] GPU recovery disabled.
[25352.378615] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, signaled seq=1691807, emitted seq=1691809
[25352.378702] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* Process information: process vulkan-smoketes pid 28192 thread vulkan-smoketes pid 28192
[25352.378706] [drm] GPU recovery disabled.
[25362.618728] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, signaled seq=1691807, emitted seq=1691809
[25362.618811] [drm:amdgpu_job_timedout [amdgpu]] *ERROR* Process information: process vulkan-smoketes pid 28192 thread vulkan-smoketes pid 28192
[25362.618814] [drm] GPU recovery disabled.

leocodeninja, 2019-11-16:
I have the same problem but with a Ryzen 7 3700U

KevinMX, 2019-12-30:
Same issue here on Radeon 520.

JacobHeAMD, 2020-08-19:
Close it since it's a request to KMD instead of AMDVLK.

closed 2020-08-19 by JacobHeAMD