GPUOpen-Drivers/AMDVLK #92: VK_ERROR_OUT_OF_HOST_MEMORY on AMD Radeon R5 M430

Niel66, 2019-06-06:
Hi,
After I install AMDVLK I got a error message as follows:

`niel@hp-notebook-15-ba017ax:~$ vulkaninfo
ERROR: [Loader Message] Code 0 : /usr/lib/i386-linux-gnu/amdvlk32.so: wrong ELF class: ELFCLASS32
/build/vulkan-tools-1.1.106.0~rc2/vulkaninfo/vulkaninfo.c:923: failed with VK_ERROR_OUT_OF_HOST_MEMORY
niel@hp-notebook-15-ba017ax:~$ `

JacobHeAMD, 2019-07-18:
Please check if amdgpu kernel module is inserted with "lsmod | grep amdgpu".

Niel66, 2019-07-18:
Yes it is loaded

JacobHeAMD, 2019-07-18:
Please check if there is any error with "dmesg", and you can debug to find out why it reports out-of-memory if there is no error message in kernel.

Niel66, 2019-07-18:
Okay. I don't know what happened but now it is showing

ERROR: [Loader Message] Code 0 : /usr/lib/i386-linux-gnu/amdvlk32.so: wrong
ELF class: ELFCLASS32
amdgpu_device_initialize: DRM version is 2.50.0 but this driver is only
compatible with 3.x.x.
amdgpu_device_initialize: DRM version is 2.50.0 but this driver is only
compatible with 3.x.x.
WARNING: [Loader Message] Code 0 : terminator_CreateInstance: Failed to
CreateInstance in ICD 0.  Skipping ICD.
Cannot create Vulkan instance.
/build/vulkan-tools-35Ffza/vulkan-tools-1.1.101.0+dfsg1/vulkaninfo/vulkaninfo.c:884:
failed with VK_ERROR_INCOMPATIBLE_DRIVER

JacobHeAMD, 2020-08-19:
Please check if there is other kernel module driver loaded, such as radeon? The log shows that the device node the vulkan driver opened is not created by amdgpu.

Niel66, 2020-08-20:
I checked using `lshw -c video`
`*-display `                
       `description: VGA compatible controller`
       `product: Mullins [Radeon R4/R5 Graphics]`
       `vendor: Advanced Micro Devices, Inc. [AMD/ATI]`
       `physical id: 1`
       `bus info: pci@0000:00:01.0`
       `version: 45`
       `width: 64 bits`
       `clock: 33MHz`
       `capabilities: pm pciexpress msi vga_controller bus_master cap_list rom`
       `configuration: driver=radeon latency=0`
       `resources: irq:41 memory:e0000000-efffffff memory:f0000000-f07fffff ioport:5000(size=256) memory:f0e00000-f0e3ffff memory:c0000-dffff`
  `*-display`
       `description: Display controller`
       `product: Sun XT [Radeon HD 8670A/8670M/8690M / R5 M330 / M430 / Radeon 520 Mobile]`
       `vendor: Advanced Micro Devices, Inc. [AMD/ATI]`
       `physical id: 0`
       `bus info: pci@0000:01:00.0`
       `version: 83`
       `width: 64 bits`
       `clock: 33MHz`
       `capabilities: pm pciexpress msi bus_master cap_list rom`
       `configuration: driver=radeon latency=0`
       `resources: irq:46 memory:d0000000-dfffffff memory:f0d00000-f0d3ffff ioport:4000(size=256) memory:f0d40000-f0d5ffff`
`

Zakhrov, 2020-08-24:
> I checked using `lshw -c video`
> `*-display `
> `description: VGA compatible controller`
> `product: Mullins [Radeon R4/R5 Graphics]`
> `vendor: Advanced Micro Devices, Inc. [AMD/ATI]`
> `physical id: 1` [...]

You are using the radeon kernel module. You can switch it to the amdgpu kernel module by adding the following to your kernel parameters: `amdgpu.si_support=1 amdgpu.cik_support=1 radeon.si_support=0 radeon.cik_support=0`

closed 2021-03-05 by jinjianrong