GPUOpen-Drivers/AMDVLK #93: Missing dependcies for build on FC30 kernel 5.1.6

cmdrogogov, 2019-06-06:
cmake3 step in Fedora30 results in errors;

-- Detected ICD_PAL_CLIENT_MAJOR_VERSION is 501
-- Detected ICD_LLPC_CLIENT_MAJOR_VERSION is 24
-- Detected ICD_GPUOPEN_CLIENT_MAJOR_VERSION is 38
**-- Could NOT find Z3: Found unsuitable version "0.0.0", but required is at least "4.7.1" (found /usr/lib64/libz3.so)**
**-- Could NOT find LibXml2 (missing: LIBXML2_LIBRARY LIBXML2_INCLUDE_DIR)** 
-- Native target X86 is not selected; lli will not JIT code
-- Threads enabled.
-- Doxygen disabled.
-- Go bindings disabled.
**-- Could NOT find OCaml (missing: OCAMLFIND OCAML_VERSION OCAML_STDLIB_PATH) 
-- Could NOT find OCaml (missing: OCAMLFIND OCAML_VERSION OCAML_STDLIB_PATH)** 
-- OCaml bindings disabled.
**-- Could NOT find Python module pygments
**-- Could NOT find Python module pygments.lexers.c_cpp
**-- Could NOT find Python module yaml**
-- LLVM host triple: x86_64-unknown-linux-gnu
-- LLVM default target triple: x86_64-unknown-linux-gnu
-- Building with -fPIC
-- Found PythonInterp: /usr/bin/python3 (found version "3.7.3") 
-- Constructing LLVMBuild project information
-- Linker detection: GNU ld
-- Targeting AMDGPU
-- Failed to find LLVM FileCheck
-- git Version: v0.0.0
-- Version: 0.0.0
-- Performing Test HAVE_GNU_POSIX_REGEX -- failed to compile
-- Performing Test HAVE_POSIX_REGEX -- success
-- Performing Test HAVE_STEADY_CLOCK -- success
-- Configured LLPC compiler options for GCC.
-- LLVM execuables: /vulkandriver/drivers/xgl/builds/Release64/llvm/./bin
-- LLVM libraries: /vulkandriver/drivers/xgl/builds/Release64/llvm/./lib
llvm-config: error: component libraries and shared library

resolved by installing these packages in addition to the ones indicated in the instructions;

`sudo dnf install libxml2* python3-pygments* yaml* ocaml* llvm-devel z3* libz3*`

I probably don't need all those packages, but it should be clear I don't really know what I'm doing.

The cmake3 step still fails because the following are missing;

llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMCore.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMAMDGPUUtils.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMAMDGPUAsmPrinter.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMAMDGPUInfo.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMAMDGPUDesc.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMAMDGPUAsmParser.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMBitReader.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMObject.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMProfileData.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMAnalysis.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMBitWriter.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMTransformUtils.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMAggressiveInstCombine.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMInstCombine.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMScalarOpts.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMTarget.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMCodeGen.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMAsmPrinter.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMAsmParser.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMIRReader.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMLinker.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMVectorize.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMInstrumentation.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMipo.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMSelectionDAG.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMGlobalISel.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMMIRParser.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMAMDGPUCodeGen.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMAMDGPUDisassembler.a
llvm-config: error: missing: /vulkandriver/drivers/xgl/builds/Release64/llvm/lib/libLLVMPasses.a

I tried getting around this by enabling the che-mesa and che-llvm copr repos 
https://copr.fedorainfracloud.org/coprs/che/

but this did not seem to help me.

Therefore, it appears the repo sync it not grabbing the appropriate llvm libraries.

Hope this helps someone.

tomkv, 2019-06-06:
These libraries are built as a part of the AMDVLK build process.

However, linking them requires a lot of memory (as in 32GB without swap is not enough). Check your build log whether you don't have linked failures there.

Otherwise, it currently does not build on F30, due to gcc9 changes. See #89.

cmdrogogov, 2019-06-07:
I was able to install the F29 package from your copr repo by editing it thus (changes in bold);

`[tkov-amdvlk]
name=Copr repo for amdvlk owned by tkov
baseurl=https://copr-be.cloud.fedoraproject.org/results/tkov/amdvlk/**fedora-29-**$basearch/
type=rpm-md
skip_if_unavailable=True
gpgcheck=1
gpgkey=https://copr-be.cloud.fedoraproject.org/results/tkov/amdvlk/pubkey.gpg
repo_gpgcheck=0
enabled=1
enabled_metadata=1
`

Re; the link failure, still nowhere on that but as it won't compile anyway, it's a lot of work for no result.

Thanks for your efforts!

closed 2019-06-07 by cmdrogogov