GPUOpen-Drivers/AMDVLK #95: wsa is not listed in default.xml

mcoffin, 2019-06-14:
If you build following the instructions in `README.md`, and enable wayland support, then it won't download the repository, as it's not listed in `default.xml`, so `repo` will not download it.

It also makes it quite hard to understand what version of `wsa` one should be using.

Can you please add the correct version to `default.xml`? I would just open a PR, but I'm not sure that it should be using GPUOpen-Drivers/wsa@master .

Thanks in advance.

JacobHeAMD, 2019-06-14:
Sorry, looks like readme is left out to be updated. wsa is deprecated and it is integrated into Pal already. Will update the readme.

mcoffin, 2019-06-16:
@JacobHeAMD It would probably be worth adding a warning (and not building WSA) if the user enables the `BUILD_WSA` option in `cmake` for `xgl`.

Cheers.

closed 2019-08-06 by mcoffin