GPUOpen-Drivers/AMDVLK #99: How to debug hangs?

godlikepanos, 2019-06-23:
Hi,

When running some of my demos I get the following on dmesg:
```
[drm:amdgpu_dm_atomic_commit_tail [amdgpu]] *ERROR* Waiting for fences timed out.
[drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, but soft recovered
```

Is there a way to find the cause of these hangs? Is there another way to debug them? 

Thank you in advance.

PS: RADV has a feature where it dumps shader information related to the hangs.

godlikepanos, 2019-07-01:
@JacobHeAMD I'm sorry for the poke. 

The hangs I'm experiencing happen only on AMDVLK. RADV and nVidia don't have similar issues. Would it be easier to share a repro-case?

I thought of asking for a way to debug myself instead of asking you guys to waste time debugging.

jinjianrong, 2019-07-01:
@godlikepanos we have some scripts to dump GPU hang status based on UMR, will try to get these open-sourced.

godlikepanos, 2019-07-01:
That would be very helpful. Thank you!