GPUOpen-Drivers/AMDVLK #101: Navi10 support on launch day?

oscarbg, 2019-07-03:
Hi,
just asking if we can expect an AMDVLK release on Navi launch day (7 july) or shortly after with support for 5700XT..

jinjianrong, 2019-07-04:
We are doing IP scrub on the source code. Might be a little bit later than launch day.

oscarbg, 2019-07-05:
thanks for status update..

Degerz, 2019-07-08:
@jinjianrong Additionally, can we also have a documentation of the Navi ISA and especially the 3D/Compute register reference guide ? (the [last](http://developer.amd.com/wordpress/media/2013/10/CIK_3D_registers_v2.pdf) time I remember AMD releasing register reference guide was the sea islands generation)

oscarbg, 2019-07-09:
+1 on Navi ISA ..

closed 2019-07-17 by oscarbg

Degerz, 2019-08-02:
@jinjianrong Thanks for releasing [documentation](https://gpuopen.com/compute-product/amd-rdna-1-0-instruction-set-architecture/) on the Navi ISA. It would be very nice if we got the register reference guide for Navi as well ...