GPUOpen-Drivers/AMDVLK #104: Segfault on release build

Chryseus, 2019-07-16:
I've been having an issue with a segmentation fault occurring in amdvlk64.so whenever amdvlk is used, it was previously working so the issue is either due to an update to amdvlk or another package, both the latest git build and v-2019.Q3.1 have this problem, I will try older versions when I get some time to see if I can narrow it down.

Building it with -DCMAKE_BUILD_TYPE=Debug makes it work without any noticable problem, I'm using the Arch [amdvlk-git](https://aur.archlinux.org/packages/amdvlk-git/) PKGBUILD, the following was given when testing it with vkcube.
https://paste.ee/p/K0Ozw

I'll try get some more info on the problem soon, any suggestions would be welcome.

Chryseus, 2019-07-16:
After much mucking around it seems to be caused by compiler optimization, I went from:

> CFLAGS="-mtune=native -march=native -O2 -pipe -fno-plt"
> CXXFLAGS="-mtune=native -march=native -O2 -pipe -fno-plt"
> LDFLAGS="-Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now"

to

> CFLAGS="-O2 -pipe -fno-plt"
> CXXFLAGS="-O2 -pipe -fno-plt"
> LDFLAGS="-Wl,-O0,--sort-common,--as-needed,-z,relro,-z,now"

closed 2019-07-16 by Chryseus

redchenjs, 2019-07-16:
https://github.com/GPUOpen-Drivers/AMDVLK/issues/50#issuecomment-472315431

Flakebi, 2021-07-09:
This issue is at last fixed in 2021.Q3.1. Compiling with avx and avx2 support is working now.