GPUOpen-Drivers/AMDVLK #105: Artifacting on Navi10

mcoffin, 2019-08-05:


mcoffin, 2019-08-06:
As a starting place, is the expected (i.e. y'all know it's not ready yet), or am I experiencing an unexpected issue?

VadikLeshy, 2019-08-07:
What do the artifacts look like?
Can [this](https://github.com/GPUOpen-Drivers/AMDVLK/issues/106) be related?

mcoffin, 2019-08-07:
@VadikLeshy They look slightly different from that. In fact, they're almost identical to what I was seeing with mesa 19.2 git RADV up until a few weeks back.

It could be similar though, I'll try to get a screenshot when I get home tonight.

MadByteDE, 2019-08-30:
I can confirm this behaviour for all games I tested using DXVK(1.3.2). [Here are some screenshots](https://imgur.com/a/3DOWATz).
I don't have any artifacting when using the vulkan-radeon-git driver. 

[Hardware Infos](https://docs.google.com/document/d/1GUX-GAKn4-kLSwy8PBghxg7eUox_D6FuNSXkaVK7udg/edit?usp=sharing)

## I'm using 
- navi10 firmware (26.08.2019)
- latest amd-staging-drm-next
- mesa-git
- latest amdvlk-git
- latest llvm-git
- latest xf86-video-amdgpu-git

aqxa1, 2019-08-31:
I have the same problem with AMDLVK. I'll also add that extracting the AMDGPU-PRO Vulkan driver (19.30-855429) and running it on top of the Mesa stack (I haven't tested the full pro stack), results in the same bug. So there may be something in common there.

VadikLeshy, 2019-09-01:
On my APU Ryzen 5 2400G (Raven Ridge) I have a problem very similar (judging by the screenshots) to the @MadByteDE 's one.

https://github.com/GPUOpen-Drivers/AMDVLK/issues/106

mcoffin, 2019-09-05:
@aqxa1 mesa has fixed this if you pull from the latest git master.

I still experience the issue with AMDVLK. It's a real bummer to be honest. I have yet to find a single person for whom the AMDVLK driver is currently working on their navi10 card

mcoffin, 2019-09-05:
> [Here are some screenshots](https://imgur.com/a/3DOWATz).

Yep. This is the exact problem that I'm having as well @MadByteDE

mcoffin, 2019-09-05:
It's also worth noting that `2019.Q3.5` works for me for my RX 590, but not the 5700XT.

mcoffin, 2019-09-06:
Just tried again with `linux-5.3rc7`, and even on that `amdgpu` driver I'm still getting the same issue(s).

mcoffin, 2019-09-06:
@MadByteDE @UnicornsOnLSD Do either of you have the 5700 non-XT? Wondering if this is isolated to the 5700 XT or gfx10_0 as a whole

UnicornsOnLSD, 2019-09-06:
@mcoffin I have the 5700 XT

mcoffin, 2019-09-07:
I hope at some point we can get a confirmation from one of the AMD devs that they can at least duplicate this problem

mcoffin, 2019-09-07:
After a lot of work patching `llpc` to use upstream LLVM 9 (with only the needed intrinsics patches from GPUOpen-Drivers/llvm), I can now confirm that all my work was for naught because the issue still persists.

This leads me to believe that one of the following three options is the case.

1. the AMD devs aren't testing against the open source `amdgpu` driver, and there's an interaction issue there.
2. There's a bug in the gfx10-specific code
3. There is something wrong with NGG for AMDVLK, because it doesn't seem to obey the settings option to disable it, so there may be something funky going on there.

UnicornsOnLSD, 2019-09-07:
@mcoffin Thanks for having a look, I hope AMD looks into this soon and confirms that it is a bug.

mcoffin, 2019-09-07:
Honestly, just hearing back from the AMD devs saying that they can replicate the problem would go a LONG way at this point. I've spent literally every waking second of the last two days trying to fix it, and while I learned a *ton* about GPU driver design, and the whole AMDVLK ecosystem in general, I still wasn't able to get it working.

I did however manage to update the LLVM version in AMDVLK to be rebased off of the latest LLVM, which is kinda nice.

mcoffin, 2019-09-07:
Also, if anyone here knows what this kind of corruption even LOOKS like, I'd appreciate any info you can guess at since I'm trying to fix this right now.

UnicornsOnLSD, 2019-09-07:
Here are the images from my issue, GTA V's artifacting is fullscreen, I just had the graphics settings done incorrectly.
[image]
[image]

UnicornsOnLSD, 2019-09-07:
Judging by [this website](http://www.playtool.com/pages/artifacts/artifacts.html), it kind of looks like VRAM artifacting but that's a pretty uneducated guess. Good luck on your endeavours, @mcoffin :)

mcoffin, 2019-09-08:
Honestly looks a bit like

jinjianrong, 2019-09-09:
Thanks all for reporting the issue, We are checking it.

qiaojbao, 2019-09-11:
I checked the upstream drivers for Navi10, it seems works fine on my ubuntu 18.04.2 system.
I test the vulkaninfo, vkcube, and steam games, like F1 2017, Mad Max, and The Talos Principle.
I will try the DXVK later.

**System:**
Processor:	Intel(R) Core(TM) i7-4790K CPU @ 4.00GHz
Motherboard:	Z77 Extreme4
RAM:		G.Skill DDR3-1600 12GB
Kernel:		v5.3-rc8
AMDVLK:	amdvlk_2019.Q3.5
libdrm:		libdrm-amdgpu1 2.4.99+git1909070630.10cd9c~oibaf~b
libllvm:	        libllvm9 1:9~+rc2-1~exp1~oibaf~b

**Steps:**
1 install ubuntu18.04.2
  apt-get update
  apt-get dist-upgrade
2 Copy Navi10 firmware into /lib/firmware/amdgpu, you can get firmware from
https://people.freedesktop.org/~agd5f/radeon_ucode/navi10/
  Install kernel, you can get kernel v5.3rc8 at 
https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.3-rc8/
  install AMDVLK driver, get it at 
https://github.com/GPUOpen-Drivers/AMDVLK/releases/download/v-2019.Q3.5/amdvlk_2019.Q3.5_amd64.deb
  Reboot system and choose 5.3rc8 kernel.
3 Add PPA and update the latest libllvm, libdrm and so on.
add-apt-repository ppa:oibaf/graphics-drivers
  apt-get update
  apt-get dist-upgrade (will install libllvm9, update libdrm-amdgpu...)
  Reboot system.

mcoffin, 2019-09-11:
I am also running those running nearly those identical specs @qiaojbao. What's different with that libdrm version (`libdrm-amdgpu1`), compared to just `libdrm-2.4.99`?

That's the only difference I see in our setups at a first glance. I'll be around tonight to do some more testing (possibly with the binary packages you mentioned just to do a complete test).

Lastly, the libllvm version shouldn't matter, given that (by default, without overriding `XGL_LLVM_SRC_PATH`) `AMDVLK` should be using it's own LLVM version, yes? Some of `llpc` doesn't even compile with the upstream LLVM due to the validator not liking the pre-inlined functions.

jinjianrong, 2019-09-11:
libllvm is only used by MesaGL, not by AMDVLK. 
AMDVLK uses its own LLVM version in amd-vulkan-dev or amd-vulkan-master branch

qiaojbao, 2019-09-14:
Thanks all for this issue.