GPUOpen-Drivers/AMDVLK #106: Artifacts with Raven Ridge

VadikLeshy, 2019-08-07:
There is heavy noises in every vulkan application (tested vkmark, TABS with DXVK, Path of Exile with DXVK) on AMDVLK 2019.Q3.3. RADV on this system works fine.

Screenshots: https://imgur.com/a/uSi0aGy

APU: Ryzen 5 2400G
RAM: DDR4 Goodram Iridium CL16 [2800 Mhz] 8Gb x 2
Motherboard: MSI AM4 X470 Gaming Pro
OS: Arch Linux x86_64
LLVM: 8.0.1
Linux 5.2.6

jinjianrong, 2019-08-08:
Is there another  DGPU on your system?

VadikLeshy, 2019-08-08:
Nope. There is only a Raven Ridge APU, doing a work of both CPU and GPU.

dzatenatskiy, 2019-09-02:
Same trouble.
APU: Ryzen 5 3400G
ASROCK DESKMINI A300
OS: Ubuntu 18.04.3
LLVM: 7.0.0
Linux: 5.0.0-25-generic
Mesa 19.3.0

VadikLeshy, 2019-10-28:
Trouble fixed for me with newer version of AMDVLK (2019.Q4.1).

OS: Arch Linux x86_64
LLVM: 9.0.0
Linux 5.3.7

closed 2019-10-28 by VadikLeshy