GPUOpen-Drivers/AMDVLK #109: Cemu Vulkan 1.16.0 WIP crashes on Vega (not on Polaris) with Zelda BOTW (but works with Radeon GPU Profiler service enabled)

oscarbg, 2019-08-24:
Hi,
don't know if you are aware but for a month all Cemu Vulkan preview builds (1.16WIP and only for patreons currently) cause crashes upon "new game" on Zelda breath of the wild on Vega GPUs on latest 19.8.1 driver as well as latest AMDVLK driver shared here..
what is more strange is that we don't have that artifact on Polaris hardware.. (some youtube videos show Vega VII (Vega 20) isn't affected also)..

I attach log with WIP7 altough latest WIP9 version fails too:
[log - copia (3).txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/3536708/log.-.copia.3.txt)
error seems:
[09:30:23] Failed to create graphics pipeline. Error -4
what's even more strange is that if you install latest Radeon GPU Profiler 1.6 and start Radeon Developer Panel and press "connect" so it starts the service (and an RGP OSD appears on all Vulkan apps) the Cemu Vulkan works correctly on Vega..

Hope AMD Vulkan guys gave a look at it as it affects it's latest Windows driver (2.0.99), AMDVLK open source driver (2.0.104) and also the Vulkan driver shipped with AMDGPU Pro 19.30 (shows as version 2.0.100)

JacobHeAMD, 2020-08-19:
@oscarbg , we don't have the environment to reproduce this issue, can you help to debug a little bit to find out why it's failed to create the pipeline?

jinjianrong, 2021-03-05:
Please re-open the issue if the issue is still there

closed 2021-03-05 by jinjianrong