GPUOpen-Drivers/AMDVLK #111: Bug/crash running Direct Display (KHR_display) programs without Xorg..

oscarbg, 2019-09-07:
Hi,
I tested running 3 programs with KHR_display support and they fail..
only Vulkan program running from console is vulkaninfo works and reports correct info..

of course I'm on terminal and without Xorg server running (on grub I attach 3 to boot in terminal/console mode)..
I tested both latest v-2019.Q3.5 open release and using latest AMDVLK from AMDGPU Pro 19.30 (http://vulkan.gpuinfo.org/displayreport.php?id=6631#device)
(Pro version (driverversion : 2.0.100 ,apiversion : 1.1.113))
I'm on Ubuntu 19.04 and tested using latest included Ubuntu 19.04 kernel 5.0.xx and also latest 5.3RC from ubuntu kernel ppa (https://kernel.ubuntu.com/~kernel-ppa/mainline)just in case some AMDGPU DRM bug..
note both Vulkan drivers correctly work in Xorg (vulkaninfo,vkcube,etc..)
I tested this 3 programs (all I know :-( ):
1) official Khronos vkcube (https://github.com/KhronosGroup/Vulkan-Tools) compiled with cmake option:
CUBE_WSI_SELECTION STREQUAL ="DISPLAY"
2) vkcube from https://github.com/krh/vkcube/ which recently added KHR_display support (https://github.com/krh/vkcube/commit/504059b6af0de234981bbdb0656702af24afb316)
3) retroarch latest version.. 
For that you need to modify $HOME/.config/retroarch/retroarch.cfg with:
video_context_driver = "khr_display"
video_driver = "vulkan"

2 notes:
a) I verified this 3 programs work in console mode using KHR_display in all other Vulkan drivers I can test: with RADV, with Anvil and with NV propietary driver (I have a PC with 3 GPUs: Vega, Intel HD530 and NV Titan GPU)
b) I even tested using root mode (sudo ...) in case was an privilege/permissions issue..

Note this was a first step in my testing of supposed HDR support in direct mode of AMDVLK since latest versions..
Hope you can help..

so finally I run strace from example 2) running 
   ./build/vkcube -m khr -k 0:0:0
it ends with both 2 AMDVLK drivers (open and propietary) with a killed by SIGSEGV ..
segment violation (core generated):
I attach screenshots of strace before crash:

note I check different info reported by KHR_display running all below and all data reported seems to make sense:
To list displays (monitors) :
   ./build/vkcube -m khr
To list modes for display 0 :
   ./build/vkcube -m khr -k 0
Finally to list planes for display 0 in mode 0 :
   ./build/vkcube -m khr -k 0:0
You can then start the vkcube on the display 0 in mode 0 on plane 0 :
   ./build/vkcube -m khr -k 0:0:0

JacobHeAMD, 2019-09-09:
Hi @oscarbg , it's working on my platform, Ubuntu18.04 with kernel 5.0.0-27, Vega10 only, libdrm 2.4.0. Probably it's caused by libdrm or multiple GPU. Could you please help to debug a little bit to see where the segfault happens to? Post the callstack will be helpful.

MojoJojoDojo, 2019-09-09:
AMDVLK got so many issues when you have multiple GPUs....

oscarbg, 2019-09-11:
good to know @MojoJojoDojo..
thanks @JacobHeAMD ..
no expert on Linux debugging so maybe I'm doing to much work..
I modified krh cube sample (https://github.com/krh/vkcube/) as per first answer of:
https://stackoverflow.com/questions/77005/how-to-automatically-generate-a-stacktrace-when-my-program-crashes
basically adding a signal handler as shown there..

output using AMDVLK v-2019.Q3.5 binary from github is:
```
Error: signal 11:
./vkcube(+0x8ebb)[0x5647584cdebb]
/lib/x86_64-linux-gnu/libc.so.6(+0x43f60)[0x7f3b8995af60]
/usr/lib/x86_64-linux-gnu/amdvlk64.so(+0x2d9367)[0x7f3b86a76367]
./vkcube(+0x6697)[0x5647584cb697] ./main.c:752
./vkcube(+0x8540)[0x5647584cd540] ./main.c:1439 
./vkcube(+0x8d30)[0x5647584cdd30] ./main.c:1634 
./vkcube(main+0xcf)[0x5647584cdfd1]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb)[0x7f3b8993db6b]
./vkcube(_start+0x2a)[0x5647584c987a]
```

I add source file and line info using:
addr2line -e ./vkcube 0x65a9 0x8540 0x8d30 0x6697

/home/cosmic/445/vkcube/build2/../main.c:747
/home/cosmic/445/vkcube/build2/../main.c:1439
/home/cosmic/445/vkcube/build2/../main.c:1634
/home/cosmic/445/vkcube/build2/../main.c:752

sadly I can't get info on amdvlk64.so as maybe compiled without debugging info:

```
addr2line -e /usr/lib/x86_64-linux-gnu/amdvlk64.so 0x2d9333
??:?
```

so from this reason program fails is in AMDVLK just called before
/home/cosmic/445/vkcube/build2/../main.c:752 ..
but not always, sometimes it fails before:
on /home/cosmic/445/vkcube/build2/../main.c:747..

main.c:747 corresponds to assert(vc->image_count > 0);
main.c:752 corresponds to assert(vc->image_count <= MAX_NUM_IMAGES);
so error must be inside AMDVLK  vkGetSwapchainImagesKHR source code..

```
   vkGetSwapchainImagesKHR(vc->device, vc->swap_chain,
                           &vc->image_count, NULL);
   assert(vc->image_count > 0);
   VkImage swap_chain_images[vc->image_count];
   vkGetSwapchainImagesKHR(vc->device, vc->swap_chain,
                           &vc->image_count, swap_chain_images);

   assert(vc->image_count <= MAX_NUM_IMAGES);
```
attach modified main so you can check exact lines:
[main.zip](https://github.com/GPUOpen-Drivers/AMDVLK/files/3602680/main.zip)

I assume you would want perhaps I build AMDVLK with debug info so call stack shows exact function in AMDVLK driver?
lazy to compile AMDVLK driver in debug mode but if it's needed will find time to do it..
Open to ideas to help debugging the issue..
thanks..

JacobHeAMD, 2019-09-26:
Yes, I need the segfault call stacks. Please help to get one with debug driver. 
BTW, did you add the user into video group? You can do it with "sudo usermod -a -G video $LOGNAME" if you didn't.

yshui, 2019-09-27:
@JacobHeAMD i think vkcube crashed because vkGetSwapchainImagesKHR returned an unreasonably large image count

JacobHeAMD, 2019-09-27:
The imageCount is specified by the application, driver just store it and return it to application in vkGetSwapchainImagesKHR. I can't see any reason why it would be 32460?
BTW, Xorg is not running, right? I ask this because "DRM_IOCTL_PRIME_FD_TO_HANDLE returning -1" which means there is no permission to do this. Please try with "root" user.

JacobHeAMD, 2020-08-17:
Close it since no responding from reporter.

closed 2020-08-17 by JacobHeAMD