GPUOpen-Drivers/AMDVLK #112: SPIR-V compiler fails to build valid shader

ReaperOfSouls1909, 2019-09-18:
The attached SPIR-V vertex module fails to build on AMD's proprietary drivers and in RGA's offline compiler. To reproduce grab that file (in its binary or text form) into RGA as a vertex shader and attempt to build it. It will crash the compiler.

[true_branch.spvas.zip](https://github.com/GPUOpen-Drivers/AMDVLK/files/3627673/true_branch.spvas.zip)
[true_branch.spv.zip](https://github.com/GPUOpen-Drivers/AMDVLK/files/3627674/true_branch.spv.zip)

linqun, 2019-09-20:
reproduced with internal proprietary driver, we will try to fix it asap.

ps: it seems this test passed with AMDVLK driver. please try it if it is urgent.

ReinUsesLisp, 2019-09-21:
We managed to emit more compiler-friendly SPIR-V to workaround this issue, although there is still a path to use the attached method in some edge-cases.

linqun, 2019-09-25:
It has been fixed in our internal branch. Thank you for reporting it.

ReinUsesLisp, 2019-09-25:
Thanks for your support. [Here](https://github.com/GPUOpen-Drivers/AMDVLK/files/3651081/failed.zip) a similar shader that fails in the same way, does it pass in the branch with the fix?

linqun, 2019-09-25:
Yes, it is fixed too.

ReaperOfSouls1909, 2019-09-25:
@linqun could you hook me up with a place to report opengl issues as well? AMD has a ton of performance issues on Windows and some issues the windows driver does not i know its offtopic but amd tends to not listen about opengl we just need an ear thats all

linqun, 2019-09-26:
Do you have a simple example about the opengl issue? if yes, I can help to report it to opengl driver team.
Have you try to report issue here: https://community.amd.com/community/support-forums/drivers-software.  and due to the resource limitation, opengl performance issue may not be fixed quickly.

ReaperOfSouls1909, 2019-10-05:
i have tried there before and zero reponce and it's really hard to debug the issue cause no good opengl debugging tools left to pin point the issue but let me sum it up yuzu for nvidia runs about 45-60fps and amd no matter how high end runs 8-9fps in super mario odyssey and other titles and amd users have no other way around it.

linqun, 2019-10-08:
Could you provide some trace? like apitrace or something else. we have some concern to do investigation on open source emulators that emulate retail console games.

ReaperOfSouls1909, 2019-10-14:
i will in a second

ReaperOfSouls1909, 2019-10-14:
https://drive.google.com/file/d/1ugqdLBWDREPN0ZaP3GnNWNutV6tZKd1p/view?usp=sharing thats the apitrace uncompressed its 2.4gb

linqun, 2019-10-23:
Sorry for delay response, I did a quick check before report it to internal team.
It seems this trace is from NV driver, it uses extension ARB_compute_variable_group_size which are not supported by AMD. 
is it possible to trace it from AMD card and compare the performance between NV and AMD?
or is it possible to disable extension  ARB_compute_variable_group_size in your test?

ps: following is apitrace output in my local machine
rendered 2223 frames in 198.197 secs, average of 11.2161 fps
Could you help to add your local machine performance if you have? thanks!

ReaperOfSouls1909, 2019-10-23:
mine was 47-60fps on a 1070

ReinUsesLisp, 2019-10-23:
About `ARB_compute_variable_group_size`, it's not used in critical rendering; if it is being gracefully ignored it shouldn't have impact on what we care about in that capture (performance).

linqun, 2019-10-24:
got it. thanks!

ReinUsesLisp, 2019-12-04:
I've reduced the issue using RenderDoc. The slowdown seems to happen on a single draw, but removing that specific `glDraw*` causes it to be deferred to the next one. This makes me think we are hitting a slow path in the driver queue through state changes. [Here](https://github.com/GPUOpen-Drivers/AMDVLK/files/3923721/pipeline.zip) is the state of the pipeline when `glDraw*` is called, and here's the trace before that draw:
[image]

Don't take this as a "make X faster". If there's something we can do on our end, please let me know.

ReaperOfSouls1909, 2020-01-05:
any updates? anything else needed?

linqun, 2020-01-06:
According to internal track system, this issue has been assigned to a developer with middle priority. until now, we needn't additional info. thanks!

JacobHeAMD, 2020-08-19:
@ReaperOfSouls1909 is the issue still seen with the latest amdvlk?

ReinUsesLisp, 2020-08-19:
Do you mean the original issue about the SPIR-V module emitting an assert or the OpenGL performance issue reported after that?

ReaperOfSouls1909, 2020-08-19:
Bit confused about which issue also

JacobHeAMD, 2020-08-20:
Oops, just realize that the original issue was fixed. For the OGL issue, can you try with the latest OGL driver? The issue has been fixed internally, but not sure if it's included by the latest released OGL driver.

ReaperOfSouls1909, 2020-08-20:
Tomorrow i will test both windows and linux to see if the slow crawling opengl is fixed

Doctorwho1909, 2020-09-04:
> Oops, just realize that the original issue was fixed. For the OGL issue, can you try with the latest OGL driver? The issue has been fixed internally, but not sure if it's included by the latest released OGL driver.

Can you do me a favour let me know when the opengl fix is live in drivers out ryujinx discord and yuzu would love to know

Doctorwho1909, 2021-01-04:
any updates? at all you said the issue with opengl was fixed and would be out soon but nothing came

JakeMerdichAMD, 2021-01-04:
The crash/assert was fixed internally on 9/24/2019 and the perf issue was fixed internally on 4/29/2020 (getting some new usages of glVertex{,Array}AttribBinding() on the fast path for roughly ~75% speedup), though it wasn't fast-tracked to any releases. I sanity-checked it and both are in the current stable releases 20.11.2 and above, though I haven't checked when the fix first went public.

By all means please test this, though. If you are still seeing issues, we can look at it again.

Doctorwho1909, 2021-01-04:
https://drive.google.com/file/d/1mUOt23Bkaaui78dWC0Z30JljrgsAYoBA/view?usp=sharing this is from Ryujinx still slow and i will get one from yuzu now

Doctorwho1909, 2021-01-04:
@JakeMerdichAMD i hope those api traces help

closed 2021-03-05 by jinjianrong

ReinUsesLisp, 2021-03-26:
Just wanted to say we've confirmed that the second issue is fixed.
Thanks for fixing them!