GPUOpen-Drivers/AMDVLK #115: Question about AMDVLK and Radeon Image Sharpening

user1-github, 2019-10-01:
Hi,

I've been using AMDVLK for some of my games since early this year. Starting with one of the latest versions (around july-august) I've noticed that all games on AMDVLK started to look much better compared to the same games on RADV. In fact, they look even better than on Windows. So I'm just curious, is Radeon Image Sharpening is implemented in AMDVLK? 
After all, RIS on Windows mainly supports Vulkan and DX12 which are based on PAL like AMDVLK.

user1-github, 2019-10-12:
Has anyone else noticed this? I think the difference is very apparent, especially in games like DOOM.

aqxa1, 2019-10-13:
@user1-github Seems like it would be simple enough to take some screenshots/videos and compare the output between AMDVLK/RADV/Windows RIS OFF/Windows RIS ON to see if the differences are due to RIS, or some other difference in output. You could even throw in AMDGPU-PRO (extract the Vulkan driver from the release) to further narrow down the differences.

I don't use AMDVLK regularly because it lacks Freesync support, but the comparison would be interesting.

user1-github, 2019-10-13:
I would say a side by side comparison is not even necessary when I can easily tell the difference without it. I currently don't have a Windows installation so I can't compare it to Windows. I would have liked to check the pro-vulkan driver from AMDGPU-PRO also to see how performant is the shader compilation of the pro compiler with DXVK but it's only available on LTS distros.
At one point I thought maybe something is wrong with RADV, so I decided to try DOOM on radeonsi as well and it looked the same as on RADV. This is when I realised that nothing is wrong with the two mesa drivers and AMDVLK might have RIS because there is no other explanation to this.
That's why I opened this thread, so that hopefully I can hear what will the developers say about this.
Lately I've been using AMDVLK for pretty much all of my games even when RADV has the much better ACO compiler for smoother gameplay via DXVK, simply because I like how everything looks better on AMDVLK.

jinjianrong, 2019-10-14:
@user1-github RIS is not supported in AMDVLK

user1-github, 2019-10-14:
Interesting.. maybe the difference is also due to how the drivers handle colors because the colors also look a bit better on AMDVLK.
Anyways, thanks for the answer.

closed 2019-10-14 by user1-github