GPUOpen-Drivers/AMDVLK #117: Error compiling sessionManager.cpp

escarneiro, 2019-10-07:
Actually, I'm updating (repo sync) before compiling. In some point, it says there is some deprecated thing and claims error:

`[ 86%] Building CXX object pal/gpuopen/CMakeFiles/gpuopen.dir/src/sessionManager.cpp.o
/home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/src/sessionManager.cpp: In member function ‘DevDriver::SharedPointer<DevDriver::Session> DevDriver::SessionManager::FindOpenSession(DevDriver::SessionId)’:
/home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/src/sessionManager.cpp:302:24: error: implicitly-declared ‘DevDriver::SharedPointer<DevDriver::Session>::SharedPointer(const DevDriver::SharedPointer<DevDriver::Session>&)’ is deprecated [-Werror=deprecated-copy]
  302 |                 return pSession;
      |                        ^~~~~~~~
In file included from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/protocolSession.h:36,
                 from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/protocolClient.h:35,
                 from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/msgChannel.h:35,
                 from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/src/sessionManager.h:35,
                 from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/src/sessionManager.cpp:32:
/home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/util/sharedptr.h:207:27: note: because ‘DevDriver::SharedPointer<DevDriver::Session>’ has user-provided ‘DevDriver::SharedPointer<T>& DevDriver::SharedPointer<T>::operator=(DevDriver::SharedPointer<T>) [with T = DevDriver::Session]’
  207 |         SharedPointer<T> &operator= (SharedPointer<T> right)
      |                           ^~~~~~~~
/home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/src/sessionManager.cpp: In member function ‘void DevDriver::SessionManager::HandleReceivedSessionMessage(const DevDriver::MessageBuffer&)’:
/home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/src/sessionManager.cpp:419:45: error: implicitly-declared ‘DevDriver::SharedPointer<DevDriver::Session>::SharedPointer(const DevDriver::SharedPointer<DevDriver::Session>&)’ is deprecated [-Werror=deprecated-copy]
  419 |                     pSession = sessionIter->value;
      |                                             ^~~~~
In file included from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/protocolSession.h:36,
                 from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/protocolClient.h:35,
                 from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/msgChannel.h:35,
                 from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/src/sessionManager.h:35,
                 from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/src/sessionManager.cpp:32:
/home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/util/sharedptr.h:207:27: note: because ‘DevDriver::SharedPointer<DevDriver::Session>’ has user-provided ‘DevDriver::SharedPointer<T>& DevDriver::SharedPointer<T>::operator=(DevDriver::SharedPointer<T>) [with T = DevDriver::Session]’
  207 |         SharedPointer<T> &operator= (SharedPointer<T> right)
      |                           ^~~~~~~~
/home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/util/sharedptr.h:207:55: note:   initializing argument 1 of ‘DevDriver::SharedPointer<T>& DevDriver::SharedPointer<T>::operator=(DevDriver::SharedPointer<T>) [with T = DevDriver::Session]’
  207 |         SharedPointer<T> &operator= (SharedPointer<T> right)
      |                                      ~~~~~~~~~~~~~~~~~^~~~~
/home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/src/sessionManager.cpp:433:49: error: implicitly-declared ‘DevDriver::SharedPointer<DevDriver::Session>::SharedPointer(const DevDriver::SharedPointer<DevDriver::Session>&)’ is deprecated [-Werror=deprecated-copy]
  433 |                         pSession = sessionIter->value;
      |                                                 ^~~~~
In file included from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/protocolSession.h:36,
                 from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/protocolClient.h:35,
                 from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/msgChannel.h:35,
                 from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/src/sessionManager.h:35,
                 from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/src/sessionManager.cpp:32:
/home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/util/sharedptr.h:207:27: note: because ‘DevDriver::SharedPointer<DevDriver::Session>’ has user-provided ‘DevDriver::SharedPointer<T>& DevDriver::SharedPointer<T>::operator=(DevDriver::SharedPointer<T>) [with T = DevDriver::Session]’
  207 |         SharedPointer<T> &operator= (SharedPointer<T> right)
      |                           ^~~~~~~~
/home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/util/sharedptr.h:207:55: note:   initializing argument 1 of ‘DevDriver::SharedPointer<T>& DevDriver::SharedPointer<T>::operator=(DevDriver::SharedPointer<T>) [with T = DevDriver::Session]’
  207 |         SharedPointer<T> &operator= (SharedPointer<T> right)
      |                                      ~~~~~~~~~~~~~~~~~^~~~~
In file included from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/src/sessionManager.h:41,
                 from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/src/sessionManager.cpp:32:
/home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/util/hashMap.h: In instantiation of ‘DevDriver::Result DevDriver::HashMap<Key, Value, NumBuckets, HashFunc, EqualFunc>::Create(const Key&, Args&& ...) [with Args = {DevDriver::SharedPointer<DevDriver::Session>&}; Key = unsigned int; Value = DevDriver::SharedPointer<DevDriver::Session>; long unsigned int NumBuckets = 16; HashFunc = DevDriver::DefaultHashFunc; EqualFunc = DevDriver::DefaultEqualFunc]’:
/home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/src/sessionManager.cpp:220:70:   required from here
/home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/util/hashMap.h:226:21: error: implicitly-declared ‘DevDriver::SharedPointer<DevDriver::Session>::SharedPointer(const DevDriver::SharedPointer<DevDriver::Session>&)’ is deprecated [-Werror=deprecated-copy]
  226 |                     new(&pEntry->value) Value(Platform::Forward<Args>(args)...);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/protocolSession.h:36,
                 from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/protocolClient.h:35,
                 from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/msgChannel.h:35,
                 from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/src/sessionManager.h:35,
                 from /home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/src/sessionManager.cpp:32:
/home/eder/Projects/Desktop/vulkandriver/drivers/pal/shared/gpuopen/inc/util/sharedptr.h:207:27: note: because ‘DevDriver::SharedPointer<DevDriver::Session>’ has user-provided ‘DevDriver::SharedPointer<T>& DevDriver::SharedPointer<T>::operator=(DevDriver::SharedPointer<T>) [with T = DevDriver::Session]’
  207 |         SharedPointer<T> &operator= (SharedPointer<T> right)
      |                           ^~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [pal/gpuopen/CMakeFiles/gpuopen.dir/build.make:89: pal/gpuopen/CMakeFiles/gpuopen.dir/src/sessionManager.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:17544: pal/gpuopen/CMakeFiles/gpuopen.dir/all] Error 2
make: *** [Makefile:152: all] Error 2`

JacobHeAMD, 2019-10-08:
Could you please tell me gcc/g++ version? And which branch are you compiling, dev or master?

escarneiro, 2019-10-08:
branch is Master. My gcc:
`eder@linux-y7pc:~> rpm -qa | grep gcc
gcc9-info-9.2.1+r275327-1.1.noarch
libstdc++6-pp-gcc9-32bit-9.2.1+r275327-1.1.x86_64
gcc9-c++-9.2.1+r275327-1.1.x86_64
gcc-info-9-1.5.x86_64
libgcc_s1-32bit-9.2.1+r275327-1.1.x86_64
libgcc_s1-9.2.1+r275327-1.1.x86_64
gcc8-8.3.1+r273794-1.1.x86_64
gcc-9-1.5.x86_64
libstdc++6-pp-gcc9-9.2.1+r275327-1.1.x86_64
gcc9-9.2.1+r275327-1.1.x86_64
libstdc++6-devel-gcc8-8.3.1+r273794-1.1.x86_64
gcc8-c++-8.3.1+r273794-1.1.x86_64
gcc-c++-9-1.5.x86_64
libstdc++6-devel-gcc9-32bit-9.2.1+r275327-1.1.x86_64
libstdc++6-devel-gcc9-9.2.1+r275327-1.1.x86_64
`

My System:
`eder@linux-y7pc:~/Projects/Desktop/vulkandriver> cat /etc/*release
NAME="openSUSE Tumbleweed"
# VERSION="20190927"
ID="opensuse-tumbleweed"
ID_LIKE="opensuse suse"
VERSION_ID="20190927"
PRETTY_NAME="openSUSE Tumbleweed"
ANSI_COLOR="0;32"
CPE_NAME="cpe:/o:opensuse:tumbleweed:20190927"
BUG_REPORT_URL="https://bugs.opensuse.org"
HOME_URL="https://www.opensuse.org/"
LOGO="distributor-logo"
eder@linux-y7pc:~/Proje`

escarneiro, 2019-10-21:
Workaround:

Added the following lines
`            -Wno-error=deprecated
            -Wno-error=deprecated-copy
`

to drivers/pal/shared/gpuopen/cmake/AMD.cmake

Now it compiles fine.

shmerl, 2019-10-27:
I get the same errors when building with g++ 9.2.1 on Debian testing.

shmerl, 2019-10-27:
@escarneiro : Thanks, I was able to build it that way!

oogetyboogety, 2019-11-02:
@shmerl @escarneiro I added it to https://github.com/GPUOpen-Drivers/pal/blob/dev/shared/gpuopen/cmake/AMD.cmake#L119-L121 but it didn't work.
```
diff --git a/shared/gpuopen/cmake/AMD.cmake b/shared/gpuopen/cmake/AMD.cmake
index fd4b8dc..50762b2 100644
--- a/shared/gpuopen/cmake/AMD.cmake
+++ b/shared/gpuopen/cmake/AMD.cmake
@@ -119,6 +119,8 @@ function(amd_target_warnings name)
                 -Wall   # Enable warnings about questionable language constructs.
                 -Wextra # Enable extra warnings that are not enabled by -Wall.
                 -Werror # Turn warnings into errors.
+               -Wno-error = deprecated
+               -Wno-error = deprecated-copy
         )

```
Am i doing it the same way you did?

Thanks,
Ravi

oogetyboogety, 2019-11-02:
@shmerl @escarneiro I added it to https://github.com/GPUOpen-Drivers/pal/blob/dev/shared/gpuopen/cmake/AMD.cmake#L119-L121 but it didn't work.

diff --git a/shared/gpuopen/cmake/AMD.cmake b/shared/gpuopen/cmake/AMD.cmake
index fd4b8dc..50762b2 100644
--- a/shared/gpuopen/cmake/AMD.cmake
+++ b/shared/gpuopen/cmake/AMD.cmake
@@ -119,6 +119,8 @@ function(amd_target_warnings name)
                 -Wall   # Enable warnings about questionable language constructs.
                 -Wextra # Enable extra warnings that are not enabled by -Wall.
                 -Werror # Turn warnings into errors.
+               -Wno-error = deprecated
+               -Wno-error = deprecated-copy
         )

Am i doing it the same way you did?

Thanks

shmerl, 2019-11-03:
Yes, that's what I did as well.

escarneiro, 2019-11-03:
Yes, that is actually what I did (except by the spaces next and before '=' sign. But I guess it does not make any difference).
 Also, I needed to make a simbolic link, like 'ln -s llvm-project/llvm  llvm'

oogetyboogety, 2019-11-05:
@shmerl @escarneiro Thanks. I think spaces might actually matter
```
c++: error: =: No such file or directory
c++: error: deprecated: No such file or directory
c++: error: deprecated-copy: No such file or directory
```

JaxLinAMD, 2020-07-30:
looks like this issue is fix so close it but you can still comment on it if you have any question.

closed 2020-07-30 by JaxLinAMD