GPUOpen-Drivers/AMDVLK #119: AMDVLK doesn't work with AMD HD 8750M

ahmed-elsayed2017, 2019-10-13:
### System Specification:
**CPU:** Intel I5 4200M
**GPU:** AMD HD 8750M
**Driver:** MESA 19.2.99 LLVM 10
**AMDVLK:** v-2019.Q3.6
**DXVK:** v1.4.2
**Operating System:** Linux Mint Cinnamon 19.2 (Based on Ubuntu 18.04)

**1-** amgpu module is enabled, and Radeon is blocked from the kernel using this grub command line:
`radeon.cik_support=0 amdgpu.cik_support=1 radeon.si_support=0 amdgpu.si_support=1 modprobe.blacklist=radeon`
and I updated the grub using:
`sudo update-grub2`

**2-** I can't use amdgpu-pro driver, because it crashes my whole system after restart, and I have to uninstall it from the terminal to be able to use my system again, so that is why I always use the opensource alternative instead (MESA).

3- I make sure that I am using AMDVLK by using this command:
`VK_ICD_FILENAMES=/etc/vulkan/icd.d/amd_icd64.json`
but all Windows games looks like this:

[image]

Is my GPU supported at all?! Because amdgpu-pro drivers, and AMDVLK have never worked for me.

ahmed-elsayed2017, 2019-10-13:
**The output of** `VK_ICD_FILENAMES=/etc/vulkan/icd.d/amd_icd64.json vulkaninfo`

```
===========
VULKAN INFO
===========

Vulkan Instance Version: 1.1.70

Instance Extensions:
====================
Instance Extensions	count = 15
	VK_KHR_surface                      : extension revision 25
	VK_KHR_xcb_surface                  : extension revision  6
	VK_KHR_xlib_surface                 : extension revision  6
	VK_KHR_wayland_surface              : extension revision  6
	VK_KHR_get_physical_device_properties2: extension revision  2
	VK_KHR_get_surface_capabilities2    : extension revision  1
	VK_KHR_external_memory_capabilities : extension revision  1
	VK_KHR_device_group_creation        : extension revision  1
	VK_KHR_external_semaphore_capabilities: extension revision  1
	VK_KHR_external_fence_capabilities  : extension revision  1
	VK_EXT_debug_report                 : extension revision  9
	VK_KHR_display                      : extension revision 23
	VK_EXT_direct_mode_display          : extension revision  1
	VK_EXT_debug_utils                  : extension revision  1
	VK_EXT_display_surface_counter      : extension revision  1
Layers: count = 5
=======
VK_LAYER_VALVE_steam_fossilize_32 (Steam Pipeline Caching Layer) Vulkan version 1.1.73, layer version 1
	Layer Extensions	count = 0
	Devices 	count = 1
		GPU id       : 0 (AMD Radeon HD 8600/8700M)
		Layer-Device Extensions	count = 0

VK_LAYER_VALVE_steam_overlay_64 (Steam Overlay Layer) Vulkan version 1.1.73, layer version 1
	Layer Extensions	count = 0
	Devices 	count = 1
		GPU id       : 0 (AMD Radeon HD 8600/8700M)
		Layer-Device Extensions	count = 0

VK_LAYER_VALVE_steam_fossilize_64 (Steam Pipeline Caching Layer) Vulkan version 1.1.73, layer version 1
	Layer Extensions	count = 0
	Devices 	count = 1
		GPU id       : 0 (AMD Radeon HD 8600/8700M)
		Layer-Device Extensions	count = 0

VK_LAYER_VALVE_steam_overlay_32 (Steam Overlay Layer) Vulkan version 1.1.73, layer version 1
	Layer Extensions	count = 0
	Devices 	count = 1
		GPU id       : 0 (AMD Radeon HD 8600/8700M)
		Layer-Device Extensions	count = 0

VK_LAYER_LUNARG_standard_validation (LunarG Standard Validation Layer) Vulkan version 1.0.70, layer version 1
	Layer Extensions	count = 0
	Devices 	count = 1
		GPU id       : 0 (AMD Radeon HD 8600/8700M)
		Layer-Device Extensions	count = 0

Presentable Surfaces:
=====================
GPU id       : 0 (AMD Radeon HD 8600/8700M)
Surface type : VK_KHR_xcb_surface
Formats:		count = 2
	B8G8R8A8_UNORM
	B8G8R8A8_SRGB
Present Modes:		count = 3
	IMMEDIATE_KHR
	MAILBOX_KHR
	FIFO_KHR
[... 3069 lines cut ...]
FORMAT_ASTC_12x10_SRGB_BLOCK:
	linearTiling   FormatFeatureFlags:
		None
	optimalTiling  FormatFeatureFlags:
		None
	bufferFeatures FormatFeatureFlags:
		None

FORMAT_ASTC_12x12_UNORM_BLOCK:
	linearTiling   FormatFeatureFlags:
		None
	optimalTiling  FormatFeatureFlags:
		None
	bufferFeatures FormatFeatureFlags:
		None

FORMAT_ASTC_12x12_SRGB_BLOCK:
	linearTiling   FormatFeatureFlags:
		None
	optimalTiling  FormatFeatureFlags:
		None
	bufferFeatures FormatFeatureFlags:
		None

```

Pik-a-boo, 2019-10-14:
Known issue see issue #72 and #54. 
I think the issue is how the driver handles hybrid graphics with the intel gpu.

ahmed-elsayed2017, 2019-10-14:
These issues still open! Anyway, I will keep using RADV as usual.

MojoJojoDojo, 2019-10-20:
Looks like that even Archwiki reported on this : 
https://wiki.archlinux.org/index.php/AMDGPU_PRO.
Reverse PRIME is a solution but I have no idea how that works.

MojoJojoDojo, 2019-10-25:
If that was the issue than what this fixed?
[image]
This is a current driver change lol.

ahmed-elsayed2017, 2019-10-25:
> If that was the issue than what this fixed?
> [image]
> This is a current driver change lol.

It is not fixed, at least for me.

ahmed-elsayed2017, 2019-11-10:
This issue is fixed with AMDVLK v-2019.Q4.2

closed 2019-11-10 by ahmed-elsayed2017