GPUOpen-Drivers/AMDVLK #120: Crysis 1 with dxvk results in semi random artifacts

nicman23, 2019-10-19:
with latest and older versions of dxvk with recent mesa, starting a game (or loading a save) in Crysis 1 results artifacts.

radv on the other hand works.

jinjianrong, 2019-10-21:
@nicman23 Thanks for reporting the issue. could you capture an image to show the issue? if possible, please help capture  renderdoc trace as well.

nicman23, 2019-10-21:
pic: https://user-images.githubusercontent.com/16306097/67042704-189acb00-f131-11e9-80a5-90f98f54ebfd.jpeg

> capture renderdoc trace as well

sure no idea how to though. if you can point me to some documentation i ll do it

ashleysmithamd, 2021-05-06:
@nicman23 Could you provide a few details of the system you are using, CPU, GPU. Whether you are using the steam version of the game, if so, the version of proton. Thank-you

nicman23, 2021-05-06:
ryzen 2700, rx 580, arch linux mainline wine

i am using a disc version with all patches applied

jinjianrong, 2021-08-02:
Please try 2021.Q3.2

closed 2021-08-02 by jinjianrong

nicman23, 2021-08-02:
i do not have amd hardware atm