GPUOpen-Drivers/AMDVLK #121: amdvlk doesn't provide VK_KHR_SWAPCHAIN_EXTENSION_NAME

cryptomilk, 2019-10-22:
I'm trying to get [vkdt (darktable raw image developing extremly fast)](https://github.com/hanatos/vkdt) working. It fails with libvulkan_radeon from Mesa, though I thought to give amdvlk a try. 

However it doesn't work because VK_KHR_SWAPCHAIN_EXTENSION_NAME isn't implemented.

cryptomilk, 2019-10-22:
@hanatos FYI

cryptomilk, 2019-10-22:
Wait `VK_KHR_swapchain                    : extension revision 70` is reported by vulkaninfo, but vkCreateDevice() returns VK_ERROR_FEATURE_NOT_PRESENT.

cryptomilk, 2019-10-22:
I thought it is the missing VK_KHR_swapchain in the beginning but it turns out the feature is there. 

However it fails when requesting VK_KHR_SWAPCHAIN_EXTENSION_NAME with the device features of:

```
  VkPhysicalDeviceDescriptorIndexingFeaturesEXT idx_features = {
    .sType = VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_DESCRIPTOR_INDEXING_FEATURES_EXT,
    .runtimeDescriptorArray = 1,
    .shaderSampledImageArrayNonUniformIndexing = 1,
  };
  VkPhysicalDeviceFeatures2 device_features = {
    .sType = VK_STRUCTURE_TYPE_PHYSICAL_DEVICE_FEATURES_2_KHR,
    .pNext = &idx_features,

    .features = {
      .robustBufferAccess = 1,
      .fullDrawIndexUint32 = 1,
      .imageCubeArray = 1,
      .independentBlend = 1,
      .geometryShader = 1,
      .tessellationShader = 1,
      .sampleRateShading = 0,
      .dualSrcBlend = 1,
      .logicOp = 1,
      .multiDrawIndirect = 1,
      .drawIndirectFirstInstance = 1,
      .depthClamp = 1,
      .depthBiasClamp = 1,
      .fillModeNonSolid = 0,
      .depthBounds = 1,
      .wideLines = 0,
      .largePoints = 0,
      .alphaToOne = 1,
      .multiViewport = 0,
      .samplerAnisotropy = 1,
      .textureCompressionETC2 = 0,
      .textureCompressionASTC_LDR = 0,
      .textureCompressionBC = 0,
      .occlusionQueryPrecise = 0,
      .pipelineStatisticsQuery = 1,
      .vertexPipelineStoresAndAtomics = 1,
      .fragmentStoresAndAtomics = 1,
      .shaderTessellationAndGeometryPointSize = 1,
      .shaderImageGatherExtended = 1,
      .shaderStorageImageExtendedFormats = 1,
      .shaderStorageImageMultisample = 1,
      .shaderStorageImageReadWithoutFormat = 1,
      .shaderStorageImageWriteWithoutFormat = 1,
      .shaderUniformBufferArrayDynamicIndexing = 1,
      .shaderSampledImageArrayDynamicIndexing = 1,
      .shaderStorageBufferArrayDynamicIndexing = 1,
      .shaderStorageImageArrayDynamicIndexing = 1,
      .shaderClipDistance = 1,
      .shaderCullDistance = 1,
      .shaderFloat64 = 1,
      .shaderInt64 = 1,
      .shaderInt16 = 1,
      .shaderResourceResidency = 1,
      .shaderResourceMinLod = 1,
      .sparseBinding = 1,
      .sparseResidencyBuffer = 1,
      .sparseResidencyImage2D = 1,
      .sparseResidencyImage3D = 1,
      .sparseResidency2Samples = 1,
      .sparseResidency4Samples = 1,
      .sparseResidency8Samples = 1,
      .sparseResidency16Samples = 1,
      .sparseResidencyAliased = 1,
      .variableMultisampleRate = 0,
      .inheritedQueries = 1,
    }
  };
```

cryptomilk, 2019-10-23:
[ERR] device does not support requested feature alphaToOne
[ERR] device does not support requested feature shaderInt16

cryptomilk, 2019-10-25:
Radeon RX 470

cryptomilk, 2019-10-25:
The libvulkan_radeon.so driver from Mesa seems to support shaderInt16.

jinjianrong, 2021-03-05:
Please re-open the new if the problem is still there

closed 2021-03-05 by jinjianrong