GPUOpen-Drivers/AMDVLK #123: The Witcher 3 is corrupted with latest amdvlk

shmerl, 2019-10-27:
Just built latest amdvlk (release), and tested the Witcher 3 in Wine+dxvk (latest). I get quite a lot of corruption even on the loading screen:

[image]

knubbbe, 2019-11-08:
I got same issue with World of Warcraft Classic, running through Lutris.
Except sometimes the cursor shows fine, but game is all black.
Sometimes it's the same as @shmerl screenshot.

**Linux kernel:**
- 5.3.8-arch1-1

**amdvlk:**
- amdvlk-git r91.19a031d-1 (from Archlinux aur)

**Wine versions:**
- ge-pronoified-nofshack-4.9-x86_64
- staging 4.18
- lutris-4.16x86_64
- lutris-4.18x86_64

**DXVK version:**
- 1.44

**GPU:**
- Radeon RX Vega 10

shmerl, 2019-12-20:
This bug is gone in latest amdvlk release, so closing.

closed 2019-12-20 by shmerl