GPUOpen-Drivers/AMDVLK #124: ArmA 3 amdgpu ring gfx timeout

Chryseus, 2019-11-10:
When AMDVLK is used along with amdgpu continuous freezing of the system occurs when running ArmA 3, dmesg output:
```
[drm:amdgpu_dm_atomic_commit_tail [amdgpu]] *ERROR* Waiting for fences timed out or interrupted!
[drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, but soft recovered
```
Now I don't know if this is directly caused by this [Mesa bug](https://bugs.freedesktop.org/show_bug.cgi?id=111481) which is apparently fixed in Mesa 20 but in any case disabling AMDVLK appears to fix the problem, this problem may also affect other games using vulkan either directly or via DXVK although I've only found it so far in ArmA 3 and Project Zomboid (which was fixed by the developer).

```
amdvlk-bin 2019.Q4.2-1
OpenGL renderer string: Radeon RX 590 Series (POLARIS10, DRM 3.33.0, 5.3.8-arch1-1, LLVM 9.0.0)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 19.2.3
```

mcoffin, 2019-11-23:
It is at least POSSIBLE that this is not a driver issue. While I found it harder to reproduce with mesa, this behavior caused me to RMA my RX 590. The problem, for me, didn't present at 1080p, but when I upgraded to 1440p monitor (and a Navi card, but I experiment with both), then this started happening under low-cpu-overhead workloads (DX12/Vulkan).

1. Which 590 do you have?
2. Can you duplicate this issue using a native-linux game like the Talos Principle?
3. If yes to (2), does using the mesa driver also solve your issue there?

For my card, it came down to a bad memory module. FWIW, I cannot currently replicate this behavior on the latest release of AMDVLK, on either polaris10 or navi10.

Chryseus, 2019-11-23:
I'm using the Power Cooler Red Devil, so far other than the fixed issue with Project Zomboid which is native I've not found any other game with the same problem, I'll try see if I can find anything else that doesn't work but at the moment it seems exclusive to ArmA 3, I've not noticed any similar problems with Windows games either including ArmA 3 on Windows.

I will try update to the latest mesa git and see if that changes anything.

Chryseus, 2019-11-24:
Using mesa git didn't help so I can only conclude there is a bug in AMDVLK or there is an unusually specific problem with my GPU that only appears when using AMDVLK with ArmA 3.

jinjianrong, 2019-11-26:
@Chryseus do you run the game through steamplay proton or wine+dxvk? could you provide the version of the proton or wine+dxvk you are using? Thanks.

Chryseus, 2019-11-26:
> @Chryseus do you run the game through steamplay proton or wine+dxvk? could you provide the version of the proton or wine+dxvk you are using? Thanks.

I'm running it through proton 4.11-8 on Arch Linux, I tried building the git version of AMDVLK but the AUR package is not currently working.

jinjianrong, 2019-11-27:
@Chryseus  Looks like the game is running with OpenGL. could you double check if it is really running with AMDVLK on your system and share us your way to install and run the game?

Johnnynator, 2019-11-27:
I can reproduce it with a RX 580, `amdvlk-2019.Q4.2` and `linux 5.3.13`. I will check later with `Q4.3` and `linux 5.4`

Johnnynator, 2019-11-27:
Can't reproduce it with `amdvlk-2019.Q4.3`

Chryseus, 2019-11-27:
I can confirm the problem is fixed in 4.3

closed 2019-11-27 by Chryseus