GPUOpen-Drivers/AMDVLK #126: AMD Vulkan API with texture display issues in RPCS3 emulator

vanderleivieirainacio, 2019-11-30:
Hello people. I found a while ago about the problem that has existed in the AMD Adrenalin Edition video driver Vulkan API since the release of this driver edition (Adrenalin). Edition) are lack of textures (eg water) and background distortions (such as textures overlapping each other causing a kind of ghost effect). The problem only happens in the Adrenalin Edition (Currently version 19.11.3) drivers and not in the Crimsom Relive Edition (latest version: 17.11.4) which works the texture display perfectly. 
I tested using the latest version of the RPCS3 emulator (v0.0.7-9186-5b9df53c) and the AMD Radeon Adrenalin Edition 19.11.3 and Crimson ReLive Edition 17.11.4 (2017) video drivers so I only changed the drivers for video and not the emulator.

[image]
AMD Vulkan API with texture and background distortion issues in RPCS3 emulator (Radeon Adrenalin Edition 19.11.3)

[image]
No texture problems or background distortions in the Vulkan API (Crimsom Relive Edition 17.11.4)

[image]
AMD Vulkan API with texture and background distortion issues in RPCS3 emulator (Radeon Adrenalin Edition 19.11.3)

[image]
No texture problems or background distortions in the Vulkan API (Crimsom Relive Edition 17.11.4)

[image]
AMD Vulkan API with texture and background distortion issues in RPCS3 emulator (Radeon Adrenalin Edition 19.11.3)

[image]
No texture problems or background distortions in the Vulkan API (Crimsom Relive Edition 17.11.4)

[image]
AMD Vulkan API with texture and background distortion issues in RPCS3 emulator (Radeon Adrenalin Edition 19.11.3)

[image]
No texture problems or background distortions in the Vulkan API (Crimsom Relive Edition 17.11.4)

[image]
AMD Vulkan API with texture and background distortion issues in RPCS3 emulator (Radeon Adrenalin Edition 19.11.3)

[image]
No texture problems or background distortions in the Vulkan API (Crimsom Relive Edition 17.11.4)

[image]
AMD Vulkan API with texture and background distortion issues in RPCS3 emulator (Radeon Adrenalin Edition 19.11.3)

[image]
No texture problems or background distortions in the Vulkan API (Crimsom Relive Edition 17.11.4)

[image]
AMD Radeon Adrenalin Edition 19.11.3 video driver recently used for testing.

[image]
AMD Radeon Crimson Relive Edition 17.11.4 (2017) video driver recently used for testing.

Note: NVIDIA video drivers currently work the same way as 2017 Crimson ReLive Edition 17.11.4. All AMD Crimson ReLive Edition video drivers edition Vulkan API works the textures perfectly. The Vulkan API problem only started with the new editions of AMD Adrenalin Edition video drivers.

JacobHeAMD, 2019-12-26:
Hi @vanderleivieirainacio , could you please help to record it with vktrace so that we can debug it?

vanderleivieirainacio, 2019-12-26:
Hi friend which program do i use to create problem trace, RadeonGPU Profile or Renderdoc? I need you to be patient with me to explain how I should use the programs to generate the vulkan trace.

Enviado do Email para Windows 10

De: Jacob He
Enviado:quinta-feira, 26 de dezembro de 2019 03:43
Para: GPUOpen-Drivers/AMDVLK
Cc:mastersonic; Mention
Assunto: Re: [GPUOpen-Drivers/AMDVLK] AMD Vulkan API with texture displayissues in RPCS3 emulator (#126)

Hi @vanderleivieirainacio , could you please help to record it with vktrace so that we can debug it?

jinjianrong, 2020-01-02:
@vanderleivieirainacio please refer to below links to capture trace with Renderdoc.
https://renderdoc.org/docs/how/how_capture_frame.html
https://www.gamingonlinux.com/wiki/Using_Renderdoc_capture

kd-11, 2020-01-06:
I'm fairly certain this issue was fixed on the emulator side. The cause of the difference is clamping of writes to gl_FragDepth in FS. Old drivers behaved like OpenGL and clamped the output so that values greater than 1 clamped and did not get clipped, new behavior does not do this. I could never find any explicit definitive statements about what is supposed to happen in such a situation, but all vulkan drivers seem to allow range extend and clipping behavior so we decided to just assume that is what spec requires.

vanderleivieirainacio, 2020-01-06:
kd-11 Thank you so much friend! I just checked and the Vulkan API is working 100% for me! Thank you very much for your attention and sorry for all the inconvenience!

closed 2020-01-06 by vanderleivieirainacio