GPUOpen-Drivers/AMDVLK #127: Visual corruption on Ryzen 5 3500U, Vega 8 / Mesa LLVM 9 works

datenwolf, 2019-12-02:
System Info: Lenovo T495, APU Ryzen 5 3500U (mobile)

- OS: Void Linux
- Linux kernel version 5.3.14
- Xorg-7.6.5 / xorg-server-1.20.6 / xf86-video-amdgpu-19.1.0
- mesa-ati-dri-19.2.5
- amdvlk-2019.Q4.3

When testing with Sascha Willems' examples when using the AMDVLK driver, there is severe visual corruption of the rendering result. Running the same example with the Mesa drivers does work flawlessly.

Occassionally when running with the AMDVLK driver, resizing the window i.e. swapchain images may cause a system hang. If at the same time a Vulkan program is running using the Mesa driver, this hang seems to be prevented, and the recreation of the swapchain fails due to `SUBOPTIMAL_KHR`. Either way a kernel exception is logged.

[vulkaninfo.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/3912570/vulkaninfo.txt)
[dmesg.log](https://github.com/GPUOpen-Drivers/AMDVLK/files/3912573/dmesg.log)
[suboptimal.log](https://github.com/GPUOpen-Drivers/AMDVLK/files/3912574/suboptimal.log)

Screen capture video available at
https://dl.datenwolf.net/AGTDP7DYUW4PKKVR/raven_ridge_vulkan_comparisons.mkv

jinjianrong, 2019-12-03:
@datenwolf  there are 4 GPUs shown in vulkaninfo. could you disable RADV when running with AMDVLK? The other 2 seems to come from duplicated JSON files, could you check it?

datenwolf, 2019-12-03:
> could you disable RADV when running with AMDVLK?

Just checked it. Tried both disabling and completely removing from the system. No change.

> The other 2 seems to come from duplicated JSON files, could you check it?

There are no duplicated JSON files:

```
~ % strace -e openat vulkaninfo 2>&1 | egrep '.*/icd\.d/.*\.json' | sort | uniq 
openat(AT_FDCWD, "/usr/share/vulkan/icd.d/amd_icd64.json", O_RDONLY) = 3
openat(AT_FDCWD, "/usr/share/vulkan/icd.d/intel_icd.x86_64.json", O_RDONLY) = 3
openat(AT_FDCWD, "/usr/share/vulkan/icd.d/radeon_icd.x86_64.json", O_RDONLY) = 3
```

jinjianrong, 2021-03-05:
Please reopen the issue if the problem is still there.

closed 2021-03-05 by jinjianrong