GPUOpen-Drivers/AMDVLK #128: The latest stable kernel update for Ubuntu breaks AMDVLK

ahmed-elsayed2017, 2019-12-04:
The latest stable kernel update for Ubuntu breaks AMDVLK (Kernel version **5.0.0-37**)

I use now kernel 5.3.0-24 which works fine with AMDVLK, but I wanted to report this issue, because a lot of distros are based on Ubuntu, and Ubuntu uses now Kernel version 5.0.0-37

**GPU:** AMD HD 8750M 2gb of vram
**AMDVLK:** v-2019.Q4.3
**Driver:** Mesa 19.3 rc5 (kisak-mesa PPA)
**OS:** Linux Mint Cinnamon 19.3

jinjianrong, 2019-12-10:
We tried Kernel version 5.0.0-37 with Vega10, AMDVLK works well on it

ahmed-elsayed2017, 2019-12-10:
I switched back to Kernel version 5.0.0-37 to do some testing.

**1- Control:**

[image]

**2- Darksiders Genesis:** (The shadows don't look good)

[image]

3- **Blacksad:** (Some textures failed to load) 

[image]

I found out after closing Blacksad that the Ram almost full, so I decided to restart my machine, and that happened:

[image]

ahmed-elsayed2017, 2019-12-24:
Actually the other 2 issues happens with RADV as well, but the corrupted textures only happens with Control when using kernel version 5.0.0-37. I will change the title to be more accurate.

closed 2020-06-27 by Ahmed-E-86