GPUOpen-Drivers/AMDVLK #129: AMD Vulkan API Still Still Having Problems on Windows

vanderleivieirainacio, 2019-12-04:
Even after the Adrenalin video driver update 19.12.1 the Vulkan API remains broken (the video driver is broken) with texture problems and OpenGL is VERY slow on Windows 10 v1903. This problem has existed since Adrenalin video driver version 18.2.1 (2018) and so far has not been fixed!
Recalling that NVIDIA drivers work very well on this issue as well as supporting the VK_EXT_fragment_shader_interlock and GL_ARB_fragment_shader_interlock extensions that AMD strongly refuses to deploy in its video drivers since the competition (NVIDIA, Intel) already has its own video drivers. and these extensions have already become a standard for graphics drivers. AMD refuses to listen to your community because I am not the only one complaining about these things I quoted.

I tested using the same version of the RPCS3 emulator and just changed the video drivers:

[image]
Vulkan API Texture Issues (Adrenalin 19.12.1)

[image]
Vulkan API has no texture problems (Adrenalin 18.1.1)

[image]
Vulkan API Texture Issues (Adrenalin 19.12.1)

[image]
Vulkan API has no texture problems (Adrenalin 18.1.1)

Correcting here, before I said that the error started from the first Adrenalin driver but it was from the Adrenalin 18.2.1 driver and continues to the last one (19.12.1) at the moment.
Please, I am begging for your attention to this problem and to correct it!

jinjianrong, 2019-12-04:
@vanderleivieirainacio could you help capture a renderdoc trace for the issue?

vanderleivieirainacio, 2019-12-04:
jian-rong.jin@amd.com could you teach me how I can do this?

vanderleivieirainacio, 2019-12-04:
@jinjianrong 
[SonicVulkan3.log](https://github.com/GPUOpen-Drivers/AMDVLK/files/3922498/SonicVulkan3.log)

vanderleivieirainacio, 2019-12-04:
@jinjianrong Forgive me I'm still learning how to use the RenderDoc program. In a few moments I give you the logs.

vanderleivieirainacio, 2019-12-04:
@jinjianrong I tried to capture but it's my first time using RenderDoc and it gave me a huge headache.
I could not do the process you asked me, sorry but from what I saw here the problem is ejm VK_IMAGE_LAYOUT_COLOR_ATTACHMENT_OPTIMAL. As I understand it is missing a texture palette for this extension

MojoJojoDojo, 2019-12-04:
I dont think this got anything to do with shader interlock lol.
And this might be an emulator bug maybe how could you know its a driver one :? :S

vanderleivieirainacio, 2019-12-04:
@MojoJojoDojo Because I used the same version of the (current) emulator and just changed the driver using the new version and the old version. Just needed to change the video driver and not the emulator and if with the old driver has no bugs and with the new driver has bugs in the same emulator the answer is that the driver has problems

vanderleivieirainacio, 2019-12-04:
It seems to work well when the extension was VK_IMAGE_LAYOUT_PRESENT_SRC_KHR.
The change to VK_IMAGE_LAYOUT_COLOR_ATTACHMENT_OPTIMAL that is causing the problem

vanderleivieirainacio, 2019-12-04:
@jinjianrong Please you will need to see what the VK_IMAGE_LAYOUT_PRESENT_SRC_KHR extension has that the VK_IMAGE_LAYOUT_COLOR_ATTACHMENT_OPTIMAL extension does not have to try to correct this problem.

vanderleivieirainacio, 2019-12-04:
Now, if the extension VK_IMAGE_LAYOUT_PRESENT_SRC_KHR is not a replacement but an add-on and is missing from the driver, it might need to be added.

vanderleivieirainacio, 2019-12-04:
Now, if the VK_IMAGE_LAYOUT_PRESENT_SRC_KHR extension is not an old extension but an add-on and is missing from the video driver, it might need to be added.

vanderleivieirainacio, 2019-12-04:
Really seeing here I can not tell how this problem happens

jinjianrong, 2019-12-11:
@vanderleivieirainacio please refer to below links for how to  capture renderdoc trace:
https://renderdoc.org/docs/how/how_capture_frame.html 
https://www.gamingonlinux.com/wiki/Using_Renderdoc_capture

vanderleivieirainacio, 2020-01-06:
Thanks a lot guys, the problem has already been solved by the emulator!

closed 2020-01-06 by vanderleivieirainacio