GPUOpen-Drivers/AMDVLK #130: How to interpret VkPhysicalDeviceShaderFloat16Int8FeaturesKHR.float16=false vs. AMD_gpu_shader_half_float

allanmac, 2019-12-13:
How should I interpret the presence of `VkPhysicalDeviceShaderFloat16Int8FeaturesKHR.float16=false` when there is simultaneously support for `AMD_gpu_shader_half_float`?

This situation appears in my Polaris/Baffin GPU with AMDVLK.

I assume it's still safe to use float16 in a shader without enabling the feature structure?

JaxLinAMD, 2020-07-30:
@GPUOpen-Drivers/llpcreviewers can anyone answer this question?

amdrexu, 2020-07-30:
Ideally, the capability and the extension support should be consistent. So it is safe to use float16 when you detect the extension is supported or the capability flag is set. If they are mismatching, please report an issue to us. On Polaris (GFX8), float16 is partially supported with some feature and performance limitations, so it seems a defect to expose the extension on it.

closed 2021-03-05 by jinjianrong