GPUOpen-Drivers/AMDVLK #131: amdvlk build fails with gcc 9.2.1

shmerl, 2019-12-20:
Here is the error:

```
[ 87%] Building CXX object pal/gpuopen/CMakeFiles/gpuopen.dir/src/sessionManager.cpp.o
/home/user/build/amdvlk/drivers/pal/shared/gpuopen/src/sessionManager.cpp: In member function ‘DevDriver::SharedPointer<DevDriver::Session> DevDriver::SessionManager::FindOpenSession(DevDriver::SessionId)’:
/home/user/build/amdvlk/drivers/pal/shared/gpuopen/src/sessionManager.cpp:244:24: error: implicitly-declared ‘DevDriver::SharedPointer<DevDriver::Session>::SharedPointer(const DevDriver::SharedPointer<DevDriver::Session>&)’ is deprecated [-Werror=deprecated-copy]
  244 |                 return pSession;
      |                        ^~~~~~~~
In file included from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/protocolSession.h:36,
                 from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/protocolClient.h:35,
                 from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/msgChannel.h:35,
                 from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/src/sessionManager.h:35,
                 from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/src/sessionManager.cpp:32:
/home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/util/sharedptr.h:207:27: note: because ‘DevDriver::SharedPointer<DevDriver::Session>’ has user-provided ‘DevDriver::SharedPointer<T>& DevDriver::SharedPointer<T>::operator=(DevDriver::SharedPointer<T>) [with T = DevDriver::Session]’
  207 |         SharedPointer<T> &operator= (SharedPointer<T> right)
      |                           ^~~~~~~~
/home/user/build/amdvlk/drivers/pal/shared/gpuopen/src/sessionManager.cpp: In member function ‘void DevDriver::SessionManager::HandleReceivedSessionMessage(const DevDriver::MessageBuffer&)’:
/home/user/build/amdvlk/drivers/pal/shared/gpuopen/src/sessionManager.cpp:449:45: error: implicitly-declared ‘DevDriver::SharedPointer<DevDriver::Session>::SharedPointer(const DevDriver::SharedPointer<DevDriver::Session>&)’ is deprecated [-Werror=deprecated-copy]
  449 |                     pSession = sessionIter->value;
      |                                             ^~~~~
In file included from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/protocolSession.h:36,
                 from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/protocolClient.h:35,
                 from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/msgChannel.h:35,
                 from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/src/sessionManager.h:35,
                 from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/src/sessionManager.cpp:32:
/home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/util/sharedptr.h:207:27: note: because ‘DevDriver::SharedPointer<DevDriver::Session>’ has user-provided ‘DevDriver::SharedPointer<T>& DevDriver::SharedPointer<T>::operator=(DevDriver::SharedPointer<T>) [with T = DevDriver::Session]’
  207 |         SharedPointer<T> &operator= (SharedPointer<T> right)
      |                           ^~~~~~~~
/home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/util/sharedptr.h:207:55: note:   initializing argument 1 of ‘DevDriver::SharedPointer<T>& DevDriver::SharedPointer<T>::operator=(DevDriver::SharedPointer<T>) [with T = DevDriver::Session]’
  207 |         SharedPointer<T> &operator= (SharedPointer<T> right)
      |                                      ~~~~~~~~~~~~~~~~~^~~~~
/home/user/build/amdvlk/drivers/pal/shared/gpuopen/src/sessionManager.cpp:463:49: error: implicitly-declared ‘DevDriver::SharedPointer<DevDriver::Session>::SharedPointer(const DevDriver::SharedPointer<DevDriver::Session>&)’ is deprecated [-Werror=deprecated-copy]
  463 |                         pSession = sessionIter->value;
      |                                                 ^~~~~
In file included from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/protocolSession.h:36,
                 from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/protocolClient.h:35,
                 from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/msgChannel.h:35,
                 from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/src/sessionManager.h:35,
                 from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/src/sessionManager.cpp:32:
/home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/util/sharedptr.h:207:27: note: because ‘DevDriver::SharedPointer<DevDriver::Session>’ has user-provided ‘DevDriver::SharedPointer<T>& DevDriver::SharedPointer<T>::operator=(DevDriver::SharedPointer<T>) [with T = DevDriver::Session]’
  207 |         SharedPointer<T> &operator= (SharedPointer<T> right)
      |                           ^~~~~~~~
/home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/util/sharedptr.h:207:55: note:   initializing argument 1 of ‘DevDriver::SharedPointer<T>& DevDriver::SharedPointer<T>::operator=(DevDriver::SharedPointer<T>) [with T = DevDriver::Session]’
  207 |         SharedPointer<T> &operator= (SharedPointer<T> right)
      |                                      ~~~~~~~~~~~~~~~~~^~~~~
In file included from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/src/sessionManager.h:41,
                 from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/src/sessionManager.cpp:32:
/home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/util/hashMap.h: In instantiation of ‘DevDriver::Result DevDriver::HashMap<Key, Value, NumBuckets, HashFunc, EqualFunc>::Create(const Key&, Args&& ...) [with Args = {DevDriver::SharedPointer<DevDriver::Session>&}; Key = unsigned int; Value = DevDriver::SharedPointer<DevDriver::Session>; long unsigned int NumBuckets = 16; HashFunc = DevDriver::DefaultHashFunc; EqualFunc = DevDriver::DefaultEqualFunc]’:
/home/user/build/amdvlk/drivers/pal/shared/gpuopen/src/sessionManager.cpp:162:70:   required from here
/home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/util/hashMap.h:224:21: error: implicitly-declared ‘DevDriver::SharedPointer<DevDriver::Session>::SharedPointer(const DevDriver::SharedPointer<DevDriver::Session>&)’ is deprecated [-Werror=deprecated-copy]
  224 |                     new(&pEntry->value) Value(Platform::Forward<Args>(args)...);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/protocolSession.h:36,
                 from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/protocolClient.h:35,
                 from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/msgChannel.h:35,
                 from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/src/sessionManager.h:35,
                 from /home/user/build/amdvlk/drivers/pal/shared/gpuopen/src/sessionManager.cpp:32:
/home/user/build/amdvlk/drivers/pal/shared/gpuopen/inc/util/sharedptr.h:207:27: note: because ‘DevDriver::SharedPointer<DevDriver::Session>’ has user-provided ‘DevDriver::SharedPointer<T>& DevDriver::SharedPointer<T>::operator=(DevDriver::SharedPointer<T>) [with T = DevDriver::Session]’
  207 |         SharedPointer<T> &operator= (SharedPointer<T> right)
      |                           ^~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [pal/gpuopen/CMakeFiles/gpuopen.dir/build.make:89: pal/gpuopen/CMakeFiles/gpuopen.dir/src/sessionManager.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:19166: pal/gpuopen/CMakeFiles/gpuopen.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

shmerl, 2019-12-20:
A workaround:

add to: `/drivers/pal/shared/gpuopen/CMakeLists.txt`

```
if(UNIX)
...
    target_compile_options(${GPUOPEN_LIB_NAME} PRIVATE -Wno-error=deprecated-copy -Wno-deprecated-copy)
```

SteveEbey73742, 2020-01-15:
that work around did not help, still dies on the deprecated-copy error, any other things to try?

justxi, 2020-01-15:
I had a problem compiling amdvlk with gcc 9.2, I added this: https://github.com/justxi/vulkan/blob/master/media-libs/amdvlk/files/amdvlk-2019.3.5-GCC-9.patch

qiaojbao, 2020-01-22:
According to the GCC 9 release note, 
Section C++, New warnings: -Wdeprecated-copy, implied by -Wextra, warns about the C++11 deprecation of implicitly declared copy constructor and assignment operator if one of them is user-provided. 
https://gcc.gnu.org/gcc-9/changes.html

shmerl's workaround and justxi patch should all works, I tested them on my local system (ubuntu 18.04, gcc 9.2.0), and we will improve this for gcc-9 ASAP.

you can add below codes firstly in gpuopen/inc/util/sharedptr.h, line 178, to continue your build.
#if __cplusplus >= 201103L
        SharedPointer(const SharedPointer<T> &right) = default;
#endif

closed 2021-03-05 by jinjianrong