GPUOpen-Drivers/AMDVLK #134: Distorted image

pamegi8982, 2019-12-28:
Pixel positions seem to be incorrect, image becomes inrecognizable. Colors seem to be in order though. Is there any way I can fix this?
```
Linux (5.4, 5.3, 5.2)
Wayland (Sway WM)
AMD Radeon 560X based laptop
Mesa OpenGL (works well)
AMDVLK Vulkan (distorted rendering)
```
Attached images demonstrate the effect with mpv as example - image above is OpenGL and below - Vulkan. This is NOT mpv's problem as the same distortion effect takes place in every Vulkan-based rendering,  both XWayland and Wayland-native.
I've been waiting for update to solve this problem for more than half of the year and tried several kernels, but no difference spotted.
Would appreciate any help. Thank you in advance.
```
https://i.postimg.cc/zfnSLfSt/rsz-vulkan.png
```

jinjianrong, 2020-01-03:
@pamegi8982 are there two GPUs on your system? could you post the log of vkinfo?

pamegi8982, 2020-01-04:
Yes, there are two GPUs, one comes with Ryzen 5 3550H, so I guess that's would be  
`Radeon™ Vega 8`
I'm not sure how do I get log for vkinfo...

JacobHeAMD, 2020-01-06:
On Ubuntu, you can just install vulkaninfo with "apt install vulkan-tools" and then run "vulkaninfo > /tmp/1.log". For other distribution, you can get it from https://vulkan.lunarg.com/sdk/home#linux  "SDK Tarball" . You can get vulkaninfo binary from "vulkansdk-linux-x86_64-1.1.130.0\1.1.130.0\x86_64\bin" .