GPUOpen-Drivers/AMDVLK #135: Tropico 4 DXVK - texture corruption with both LLVM and proprietary compiler

aufkrawall, 2020-01-09:
amdvlk 2019.Q4.5-1:
[image]

amdvlk-pro 19.50:
[image]

RX 570 8GB
Linux 5.5-rc5
dxvk git-master 47555f1ddaf665805219456e856a1985e05289f4
Proton 4.11-11
game is D3D9 wrapped to Vulkan
no corruption with radv (LLVM & ACO)

jinjianrong, 2020-02-05:
@aufkrawall could you help capture the renderdoc trace for the issue?   thanks

aufkrawall, 2020-02-05:
It would not be possible to open it with amdvlk, as I have no more AMD card installed.

JacobHeAMD, 2020-07-16:
Close it since it's impossible to reproduce it. We can reopen it if there is anyone else see this issue and can help to capture the frame.

aufkrawall, 2020-08-03:
I'm back with an RX 480, it's still reproducible with latest proprietary amdvlk drivers (didn't try the open one).
I've finally managed to capture via RenderDoc, which was a rather painful procedure involving the 32 bit Windows build of RenderDoc. :( But: I can load the capture successfully in the 64 bit version of QRenderDoc.

Here it is:
https://drive.google.com/file/d/15S02UEI1XGL9S9Wxe706RSPZb1oKxSnS/view?usp=sharing

Created on Windows 10 2004, AMD driver 20.7.2 and latest DXVK build.

aufkrawall, 2020-08-07:
@JacobHeAMD With amdvlk open 2020.Q3.3, the game crashes at start. With amdgpu-pro 20.30-1109583, it starts but shows the corruption above on the screenshots and in the renderdoc capture.

perlfu, 2020-08-08:
I can confirm a pipeline compilation crash on starting this title with Proton 5.0 (on RX 5700).
I have attached the (first) failing pipeline.

[amdvlk-135-tropico-4.zip](https://github.com/GPUOpen-Drivers/AMDVLK/files/5044576/amdvlk-135-tropico-4.zip)

trenouf, 2020-08-24:
My commit above in AMDVLK LLPC works around the bad shader.

aufkrawall, 2020-08-24:
@trenouf Thanks for implementing a workaround for the crash issue. May I ask if what looks like texture corruption was caused by this as well?

jinjianrong, 2020-08-25:
The texture corruption issue is still under investigation, keep the issue open

aufkrawall, 2020-09-14:
Crash is gone with 2020-9-12 update, texture issue persists.

JaxLinAMD, 2020-09-15:
@aufkrawall  thanks for your feedback

JaxLinAMD, 2020-11-18:
@aufkrawall , issue is fixed in v-2020.Q4.4

aufkrawall, 2020-11-18:
Thanks. Seems it wasn't an issue on Navi? I'm currently with a 5700 XT and the older amdgpu-pro 20.40.1147286 driver doesn't show the corruption with it.

JaxLinAMD, 2020-11-19:
yes,  it wasn't an issue on Navi

closed 2021-03-05 by jinjianrong