GPUOpen-Drivers/AMDVLK #138: Timeouts when using Vulkan's subgroup operations

godlikepanos, 2020-01-20:
Hi,

Some algorithms that use subgroup operations seem to be causing timeouts on AMDVLK driver. This issue spans across multiple releases of AMDVLK but only recently I found some time to narrow it down.

Some information:
- The GPU is a Radeon RX 580
- The distro is Ubuntu 18.04
- dmesg prints `[drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, but soft recovered`
- RADV and the proprietary Windows 10 driver have no issues

One of the shaders (GLSL) that cause the timeout is attached here: [shader.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/4087652/amd_timeout.txt)
It's quite huge but the interesting part is in main().

So the issue is in the following loop:

```
UVec4 execMask = UVec4(0u);
ANKI_LOOP while((execMask & gl_SubgroupEqMask) == UVec4(0u))
{
	U32 uniformIdxOffset = subgroupBroadcastFirst(idxOffset); // Should be uniform
	const UVec4 laneMask = subgroupBallot(uniformIdxOffset == idxOffset);
	execMask |= laneMask;
	
	// ...
}
```

The `idxOffset` should be in VGPR and the `uniformIdxOffset` should be uniform and in SGPR (in theory). The `laneMask` will give a set of lanes that will be executed. This loop seems to run indefinitely and causing the timeout.

How to reproduce: Unfortunatelly I don't think I can grab a RenderDOC trace but you can build my engine pretty easily yourselves and run a sample:

```
git clone --recurse-submodules https://github.com/godlikepanos/anki-3d-engine.git anki
cd anki
mkdir build
cd ./build
cmake .. -DCMAKE_BUILD_TYPE=Release
make

cd ../samples/simple_scene
../../build/bin/simple_scene -cfg core_clearCaches 1
```

Thank you for looking into this. If you require more information please let me know.

allanmac, 2020-01-20:
Nothing leapt out at me as being incorrect in your code.

One critique, you could probably simplify this loop a little to avoid carrying around a `uvec4` when all you're doing is masking it with its own lane mask ... making it equivalent to a per-lane boolean.

```
bool is_active = true;

while (is_active) {
  
  uint       uniformIdxOffset = subgroupBroadcastFirst(idxOffset); // Should be uniform
  const bool is_match         = (uniformIdxOffset == idxOffset);

  if (is_match) {

      is_active = false;
      ...
```

FWIW, for historical reasons I try to avoid invoking subgroup operations when there are inactive lanes despite it being legal.  But as I noted above, I think what you're doing is legal.

I assume you've tried this shader on MESA/RADV (as well as the ACO optimizer) ?

godlikepanos, 2020-01-20:
@allanmac Thanks for the tip and for taking the time to look into this! I don't get timeouts with RADV with both LLVM and ACO. No timeouts on Windows 10 as well. No timeouts on Mesa Intel or Windows nVidia as well. There is a high probability for the culprit to be AMDVLK.

allanmac, 2020-01-20:
I have a very large code base of subgroup-centric code and have also noticed regressions in AMDVLK so I'm definitely leaning toward there being a problem with AMDVLK in the last 2-3 releases.

jiaolu, 2020-01-22:
this is khronos cts tests:
dEQP-VK.subgroups.ballot_broadcast.graphics.subgroupbroadcastfirst_int thats failed on the amdvlk

jiaolu, 2020-02-21:
@godlikepanos https://github.com/GPUOpen-Drivers/llpc/pull/452 has fix SubgroupBroadcastFirst issue.

godlikepanos, 2020-05-16:
Sorry for the late update. Tested the latest AMDVLK and I haven't observed any hangs. Thank you.

closed 2020-05-16 by godlikepanos