GPUOpen-Drivers/AMDVLK #139: Can't run Monster Hunter World

Elinvention, 2020-02-22:
If I try to run Monster Hunter World from Steam (proton 5.1) using AMDVLK 2020.Q1.1 by setting `VK_ICD_FILENAMES="/usr/share/vulkan/icd.d/amd_icd64.json" %command%` as launch options, a black window appears and quits soon after.
This did not happen in previous AMDVLK versions (but I'm not sure exactly which version).
My GPU is R7 370, but I tested with an RX 580 too.

Here is my vulkaninfo and the log that is generateb by `PROTON_LOG=1`.

[amdvlk_vulkaninfo.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/4239533/amdvlk_vulkaninfo.txt)
[radv_vulkaninfo.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/4239534/radv_vulkaninfo.txt)
[steam-582010.log.gz](https://github.com/GPUOpen-Drivers/AMDVLK/files/4239576/steam-582010.log.gz)

JacobHeAMD, 2020-07-16:
Did you uninstall radv before trying with amdvlk?

Elinvention, 2020-07-16:
No, I didn't. To my understanding, I just need to set `VK_ICD_FILENAMES` to choose between vulkan drivers?

JacobHeAMD, 2020-07-16:
Right. I'm just worried that VK_ICD_FILENAMES doesn't work for some cases. There is a known issue with both RADV and amdvlk loaded. Please have a try with AMDVLK driver only and paste the dmesg here. Call stacks will be very helpful if you can attach to the game with gdb.

Elinvention, 2020-08-03:
In the mean time, my GPU died. I can't test this anymore...

closed 2021-03-10 by jinjianrong