GPUOpen-Drivers/AMDVLK #141: Issue with `VK_KHR_depth_stencil_resolve`

Wunkolo, 2020-02-26:
I have been using `VK_KHR_depth_stencil_resolve` with success on my Intel and Nvidia machines, but across two windows machines now I am getting an access violation issue within `amdvlk64.dll`.

Callstack within amdvlk64.dll is:
```
>	amdvlk64.dll!00007ff9c519c8f5()	Unknown
 	amdvlk64.dll!00007ff9c519cf49()	Unknown
```

Some `vulkaninfo` snippets
```
VkPhysicalDeviceProperties:
---------------------------
        apiVersion     = 4202627 (1.2.131)
        driverVersion  = 8388735 (0x80007f)
        vendorID       = 0x1002
        deviceID       = 0x687f
        deviceType     = PHYSICAL_DEVICE_TYPE_DISCRETE_GPU
        deviceName     = Radeon RX Vega

VkPhysicalDeviceDriverProperties:
---------------------------------
        driverID           = DRIVER_ID_AMD_PROPRIETARY
        driverName         = AMD proprietary driver
        driverInfo         = 20.2.1
        conformanceVersion = 1.2.0.2
VkPhysicalDeviceDepthStencilResolveProperties:
----------------------------------------------
        supportedDepthResolveModes:
                RESOLVE_MODE_SAMPLE_ZERO_BIT
                RESOLVE_MODE_MIN_BIT
                RESOLVE_MODE_MAX_BIT
        supportedStencilResolveModes:
                RESOLVE_MODE_SAMPLE_ZERO_BIT
                RESOLVE_MODE_MIN_BIT
                RESOLVE_MODE_MAX_BIT
        independentResolveNone = true
        independentResolve     = true

```

Upon further reverse engineering of the offending `amdvlk64.dll` assembly I have found that it is trying to _iterate_ the `pDepthStencilResolveAttachment` field of `VkSubpassDescriptionDepthStencilResolve` when it should only be de-referencing a single one, causing the crash when trying to dereference index `i`

https://khronos.org/registry/vulkan/specs/1.2-extensions/man/html/VkSubpassDescriptionDepthStencilResolve.html

```
// r11 seems to be the "index"
00007FF9C519C8CE  mov         rax,r11  
// rax = r11 * 32, aka r11 * sizeof(VkAttachmentReference2)
00007FF9C519C8D1  shl         rax,5
00007FF9C519C8D5  add         rax,qword ptr [rcx+18h]
// rcx is VkSubpassDescriptionDepthStencilResolveKHR

// The above three instructions are the equivalent of:
// VkSubpassDescriptionDepthStencilResolveKHR myDepthStencilResolve
// myDepthStencilResolve.pDepthStencilResolveAttachment[r11]
---
00007FF9C519C8D9  mov         edx,dword ptr [rax+10h]  
00007FF9C519C8DC  mov         dword ptr [r8+58h],edx  
00007FF9C519C8E0  mov         edx,dword ptr [rax+14h]  
00007FF9C519C8E3  mov         dword ptr [r8+5Ch],edx  
00007FF9C519C8E7  mov         edx,dword ptr [rax+18h]  
00007FF9C519C8EA  mov         dword ptr [r8+64h],edx  
00007FF9C519C8EE  mov         edx,dword ptr [rax+14h]  
00007FF9C519C8F1  mov         dword ptr [r8+60h],edx  
// At this point, rdx is full of garbage(or 0xCC, in MSVC debug mode), and is writing garbage to r8's offsets
---
00007FF9C519C8F5  cmp         dword ptr [rax],3B9E7769h  << 💥💥💥 crashes here
// Tests the "sType" field for
// 1000241001(VK_STRUCTURE_TYPE_ATTACHMENT_REFERENCE_STENCIL_LAYOUT)
```

This is isolated code causing the issue.
```cpp
		vk::AttachmentReference2 depthResolveAttachmentRef = { 7, vk::ImageLayout::eDepthStencilAttachmentOptimal, vk::ImageAspectFlagBits::eDepth | vk::ImageAspectFlagBits::eStencil };

		vk::SubpassDescriptionDepthStencilResolveKHR depthStencilResolveInfo = {};
		depthStencilResolveInfo.depthResolveMode = vk::ResolveModeFlagBitsKHR::eSampleZero;
		depthStencilResolveInfo.stencilResolveMode = vk::ResolveModeFlagBitsKHR::eSampleZero;
		depthStencilResolveInfo.pDepthStencilResolveAttachment = &depthResolveAttachmentRef;
...
subpass.pNext = &depthStencilResolveInfo; // Removing this pNext removes the crash, but also depth/stencil resolving
```

Which works fine on Nvidia and Intel

The AMD-only "Fix" would be something silly like this, which gets the code running and with no validation errors, but with some pretty chaotic visual issues where the resolved depth/stencil texture is used.

I have **6-color attachments** in this particular instance, and the bypass requires me to have pDepthStencilResolveAttachment point to **6** redundant depth-resolve attachment references due to the fact that it iterates this variable rather than referencing it once.

```cpp
		vk::AttachmentReference2 depthResolveAttachmentRef[] = {
			{ 7, vk::ImageLayout::eDepthStencilAttachmentOptimal, vk::ImageAspectFlagBits::eDepth | vk::ImageAspectFlagBits::eStencil },
			{ 7, vk::ImageLayout::eDepthStencilAttachmentOptimal, vk::ImageAspectFlagBits::eDepth | vk::ImageAspectFlagBits::eStencil },
			{ 7, vk::ImageLayout::eDepthStencilAttachmentOptimal, vk::ImageAspectFlagBits::eDepth | vk::ImageAspectFlagBits::eStencil },
			{ 7, vk::ImageLayout::eDepthStencilAttachmentOptimal, vk::ImageAspectFlagBits::eDepth | vk::ImageAspectFlagBits::eStencil },
			{ 7, vk::ImageLayout::eDepthStencilAttachmentOptimal, vk::ImageAspectFlagBits::eDepth | vk::ImageAspectFlagBits::eStencil },
			{ 7, vk::ImageLayout::eDepthStencilAttachmentOptimal, vk::ImageAspectFlagBits::eDepth | vk::ImageAspectFlagBits::eStencil },
		};

		vk::SubpassDescriptionDepthStencilResolveKHR depthStencilResolveInfo = {};
		depthStencilResolveInfo.depthResolveMode = vk::ResolveModeFlagBitsKHR::eSampleZero;
		depthStencilResolveInfo.stencilResolveMode = vk::ResolveModeFlagBitsKHR::eSampleZero;
		depthStencilResolveInfo.pDepthStencilResolveAttachment = depthResolveAttachmentRef;
```

Wunkolo, 2020-06-22:
Any updates on this? The issue is still consistent and even the most latest driver and Vulkan 1.2 specification states that it will only be reading a _single_ `VkAttachmentReference2` from the pointer given, not `colorAttachmentCount`.

https://www.khronos.org/registry/vulkan/specs/1.2-extensions/man/html/VkSubpassDescriptionDepthStencilResolve.html

>If pDepthStencilResolveAttachment is not NULL, pDepthStencilResolveAttachment must be a valid pointer to **a** valid VkAttachmentReference2 structure

qiaojbao, 2021-03-29:
I tested some CTS cases about this with AMDVLK on Linux.

> ~/amdvlk/VulkanCTS_lnx_dev_63ad_0327$ ./deqp-vk -n dEQP-VK.renderpass2.depth_stencil_resolve*
> Test run totals:
>   Passed:        4645/16633 (27.9%)
>   Failed:        0/16633 (0.0%)
>   Not supported: 11988/16633 (72.1%)
>   Warnings:      0/16633 (0.0%)
>   Waived:        0/16633 (0.0%)

It seems no failure.

Could you give your test sample soure code for more check on this issue, please?

BTW, below is a tipcal trace log about  DepthStencilResolveAttachment.
```
 pNext:                          VkSubpassDescriptionDepthStencilResolve = 0x7ffe1d438830:
     sType:                          VkStructureType = VK_STRUCTURE_TYPE_SUBPASS_DESCRIPTION_DEPTH_STENCIL_RESOLVE (1000199001)
     pNext:                          const void* = NULL
     depthResolveMode:               VkResolveModeFlagBits = 8 (VK_RESOLVE_MODE_MAX_BIT)
     stencilResolveMode:             VkResolveModeFlagBits = 1 (VK_RESOLVE_MODE_SAMPLE_ZERO_BIT)
     pDepthStencilResolveAttachment: const VkAttachmentReference2* = 0x7ffe1d438810:
         sType:                          VkStructureType = VK_STRUCTURE_TYPE_ATTACHMENT_REFERENCE_2 (1000109001)
         pNext:                          const void* = NULL
         attachment:                     uint32_t = 1
         layout:                         VkImageLayout = VK_IMAGE_LAYOUT_DEPTH_STENCIL_ATTACHMENT_OPTIMAL (3)
         aspectMask:                     VkImageAspectFlags = 6 (VK_IMAGE_ASPECT_DEPTH_BIT | VK_IMAGE_ASPECT_STENCIL_BIT)
```

Wunkolo, 2021-04-02:
The codebase has steered away from using MSAA at all due to this bug so I do not have a minimal test case of this issue anymore in the original context but I can likely repurpose a pre-existing sample project to contain and recreate the issue again.
This issue in particular is due to the number of color attachments causing `pDepthStencilResolveAttachment` to dereference multiple structures rather than just the one. Not sure if your test cases are recreating the original situation.

qiaojbao, 2021-04-06:
I have no idea about the behavior of the driver when using an array of VkAttachmentReference2 here, it is not aligned with spec.
I had a try to modify the CTS source code to verify it.
[image]
The modified test seems no crash.
We can do more tracing on your original source code.
BTW, I think we should align with spec.

closed 2024-05-22 by jinjianrong