GPUOpen-Drivers/AMDVLK #142: ../pal/addrlib/libaddrlib.a: error adding symbols: archive has no index

commandline-be, 2020-02-26:
Performed the steps to build the AMDVLK source tree as instructed, arrived at this error in the final stage of compilation. Little clue on how to remediate this, looks like a bug.

Scanning dependencies of target xgl
[ 97%] Building CXX object icd/CMakeFiles/xgl.dir/api/vk_physical_device.cpp.o
[ 97%] Linking CXX shared library amdvlk64.so
/usr/bin/ld: ../pal/addrlib/libaddrlib.a: error adding symbols: archive has no index; run ranlib to add one
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [icd/CMakeFiles/xgl.dir/build.make:1319: icd/amdvlk64.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:18804: icd/CMakeFiles/xgl.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

qiaojbao, 2020-02-27:
@commandline-be 
Hi, could you attach the whole compile log file please.
Did you using "clang" to build amdvlk driver?

Flakebi, 2021-01-08:
I’m a little late. Anyway, it seems like you’re using clang to build.
For reference, this can be done in several ways, which have different effects with regard to the workarounds listed below:

1. Using CMake flags `-DCMAKE_C_COMPILER=clang -DCMAKE_CXX_COMPILER=clang++`
2. Setting environment variables `CC=clang CXX=clang++`
3. Adding `-DXGL_USE_CLANG=ON` to the CMake flags

First of all, using gcc as @qiaojbao said, should always just work.
For clang, sometimes workarounds are needed, depending on the distribution and clang version used. Here is a list of possible workarounds that might fix `libaddrlib.a: error adding symbols: archive has no index; run ranlib to add one` and similar errors:

1. Link with lld instead of ld: Add `-DLLVM_USE_LINKER=lld -DCMAKE_EXE_LINKER_FLAGS=-fuse-ld=lld -DCMAKE_SHARED_LINKER_FLAGS=-fuse-ld=lld` to the CMake flags
2. Add `-DXGL_USE_CLANG=ON` to the CMake flags. Note: this overwrites `CMAKE_C_COMPILER` and `CMAKE_CXX_COMPILER`
3. Set `CMAKE_AR` to `llvm-ar` and `CMAKE_RANLIB` to `llvm-ranlib`. This is done as part of `-DXGL_USE_CLANG=ON`, though might give different effects if applied on its own

Unfortunately, we do not know a single solution that is working on all linux distributions regardless of which packages are installed. We do our best to support all kinds of configurations, so if someone finds a configuration that “just works”, tell us!

As there hasn’t been a lot of activity lately and there is no real solution, I think this issue can be closed for now.

closed 2021-03-05 by jinjianrong