GPUOpen-Drivers/AMDVLK #143: Subgroup codegen bug

onatto, 2020-03-03:
The compiler generates BUFFER_ATOMIC_ADD instead of S_ATOMIC_ADD.
Reproduction:
http://shader-playground.timjones.io/28d4a04c8811e51240983b2f92f11c46

JacobHeAMD, 2020-08-19:
Hi @onatto , sorry for delay. But http://shader-playground.timjones.io/28d4a04c8811e51240983b2f92f11c46 is gone, can you post the shader instructions here?

amdrexu, 2020-08-19:
```
#version 450
layout(set=0,binding=0,std430) buffer ssbo0_ { uint bufA32[1024*1024]; };
#extension GL_KHR_shader_subgroup_vote : enable
void main()
{
    if (subgroupElect()) {
		atomicAdd(bufA32[0],1);
    }
}

; -------- Disassembly --------------------
shader main
  asic(GFX9)
  type(CS)

  s_getpc_b64   s[0:1]                                  // 000000000000: BE801C80
  s_ff1_i32_b64  s0, exec                               // 000000000004: BE80117E
  v_mbcnt_lo_u32_b32  v0, -1, 0                         // 000000000008: D28C0000 000100C1
  v_mbcnt_hi_u32_b32  v0, -1, v0                        // 000000000010: D28D0000 000200C1
  s_mov_b64     s[4:5], exec                            // 000000000018: BE84017E
  v_cmpx_eq_i32  s[6:7], s0, v0                         // 00000000001C: 7DA400F9 06868600
  s_cbranch_execz  label_0018                           // 000000000024: BF88000E
  s_mov_b64     s[6:7], exec                            // 000000000028: BE86017E
  s_ff1_i32_b64  s0, exec                               // 00000000002C: BE80117E
  s_lshl_b64    s[8:9], 1, s0                           // 000000000030: 8E880081
  s_and_b64     s[8:9], s[8:9], exec                    // 000000000034: 86887E08
  s_and_saveexec_b64  s[8:9], s[8:9]                    // 000000000038: BE882008
  s_cbranch_execz  label_0018                           // 00000000003C: BF880008
  s_mov_b32     s0, s2                                  // 000000000040: BE800002
  s_load_dwordx4  s[0:3], s[0:1], 0x00                  // 000000000044: C00A0000 00000000
  s_bcnt1_i32_b64  s6, s[6:7]                           // 00000000004C: BE860D06
  v_mov_b32     v0, s6                                  // 000000000050: 7E000206
  s_waitcnt     lgkmcnt(0)                              // 000000000054: BF8CC07F
  buffer_atomic_add  v0, v0, s[0:3], 0                  // 000000000058: E1080000 80000000
label_0018:
  s_endpgm                                              // 000000000060: BF810000
end
```

amdrexu, 2020-08-19:
@ruiling Could you take a look at this issue?

ruiling, 2020-08-24:
I checked the generated assembly, seems we don't have any bug here. we are using vector version of atomic_add, but in-fact there is only one lane executing the command, so you will still get expected behavior. So this is not a bug, but a possible enhancement. But I am not sure if there is any real user program or benchmark suffering from this sub-optimal code-generation to make it a strong reason for us to optimize for it now.

onatto, 2020-10-04:
Yes, functionally it works, sorry for calling it a bug.
I thought it was common to do wave-reduction then atomic on a single thread, noticed that the instruction seems to be removed in RDNA, I guess for architectural simplicity reasons. So agreed, don't see the point of adding the instruction just for Vega.

closed 2020-10-04 by onatto