GPUOpen-Drivers/AMDVLK #145: Assassin's Creed Syndicate broken transparency

niobium93, 2020-03-17:
Running on Arch Linux Zen 5.5.9 kernel with Wine Staging 5.4 and DXVK 1.5.5 on the AMDVLK 2020.Q1.3-1 driver on an RX 5700 XT.
Screen capture: [https://www.youtube.com/watch?v=QEF1FC6dZgM](url)
RenderDoc capture: [https://mega.nz/#!nBghiSIC!QCRnV_txLI2ASR_utuMNs8VL0IYUNRnjVhdsqYlwOJU](url)

Kron4ek, 2020-03-22:
The same issue in Kingdom Come Deliverance on RX 560.

[image]

jinjianrong, 2020-03-23:
@niobium93 There is a change for this issue in amd-gfx-gpuopen-dev branch of llvm-project. could you try it?

niobium93, 2020-03-23:
@jinjianrong I tried the dev branch and it's fixed there!
[image]

closed 2020-03-23 by niobium93