GPUOpen-Drivers/AMDVLK #146: calling vkBindImageMemory2 causes APP segfault

renguannan1, 2020-03-18:
Hi I am writing a rendering APP with AMD radeon vega 64 GPU.  When I tried to bind image object with vkBindImageMemory2() API, I got a segment fault. But when I use vkBindImageMemory() API, it worked fine.  I gave a several tries on a couple of recent both Vulkan SDK and AMDVLK versions, the case is still there.   ANY thoughts is appreciated.

Code snippet
...
    VkBindImageMemoryInfo bindInfo[1] = { };
    bindInfo[0].sType = VK_STRUCTURE_TYPE_BIND_IMAGE_MEMORY_INFO;
    bindInfo[0].image = m_vd->textureImage;
    bindInfo[0].memory = m_vd->textureImageMemory;
    bindInfo[0].memoryOffset = 0;
    bindInfo[0].pNext = NULL;
   if (vkBindImageMemory2(m_vd->device, 1, bindInfo) != VK_SUCCESS)
    {
            throw std::runtime_error("failed to allocate image memory!");
    }

...

(gdb) bt
#0  0x0000000000000000 in ?? ()
#1  0x0000555555557ddd in MainApp::createTextureImages (this=0x55555578ce70) at main.cpp:797
#2  0x000055555555757e in MainApp::initVulkan (this=0x55555578ce70) at main.cpp:343
#3  0x000055555555749c in MainApp::run (this=0x55555578ce70, argc=1 '\001', argv=0x7fffffffe538) at main.cpp:247
#4  0x0000555555556940 in main (argc=1, argv=0x7fffffffe538) at main.cpp:1367

OS Ubuntu:
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.4 LTS
Release:	18.04
Codename:	bionic

Vulkan SDK: 
amdvlk_2020.Q1.3_amd64.deb  
1.2.131.2~rc1

AMDVLK Info:
library_path: /usr/lib/x86_64-linux-gnu/amdvlk64.so
api_version: 1.2.133

AMDGPU Info:
filename:       /lib/modules/4.15.0-91-generic/updates/dkms/amdgpu.ko
version:        5.0.73.19.30

GPU Hardware:
VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Vega 10 XT [Radeon RX Vega 64]

amingriyue, 2020-03-19:
if (vkBindImageMemory2(m_vd->device, 1, **&**bindInfo) != VK_SUCCESS)

amingriyue, 2020-03-19:
correct this statement as below:
if (vkBindImageMemory2(m_vd->device, 1, &bindInfo) != VK_SUCCESS)

renguannan1, 2020-03-19:
> correct this statement as below:
> if (vkBindImageMemory2(m_vd->device, 1, &bindInfo) != VK_SUCCESS)

Thanks for your answer. I gave it a try,  but it did not pass the compiling stage.
main.cpp: In member function ‘void MainApp::createTextureImages()’:
main.cpp:799:54: error: cannot convert ‘VkBindImageMemoryInfo (*)[1]’ to ‘const VkBindImageMemoryInfo*’ for argument ‘3’ to ‘VkResult vkBindImageMemory2(VkDevice, uint32_t, const VkBindImageMemoryInfo*)’

amingriyue, 2020-03-19:
Sorry， I didn't realize it's defined as array.
can you try  &bindInfo[0]?

renguannan1, 2020-03-19:
Sure, it pass the compiling without any warning,  the segfault still occures when I run it.

amingriyue, 2020-03-19:
If vkBindImageMemory pass，that doesn't make sense. 
bt trace is same? which line leads to segfault?
m_vd is nullptr?
Are you using debug amdvlk or release amdvlk? If using debug amdvlk and segfault happens in amdvlk, we can also see which line has problem.

renguannan1, 2020-03-19:
bt trace is same
I am sure that vkBindImageMemory2() leads to segfault,  but I don't know how to switch on AMDVLK debug capability, I am digging into it, If you can help, that's great. the version is amdvlk_2020.Q1.3_amd64.deb release.
my_vd->device is not NULL,  I can print it out m_vd->device is 0x559ca64d4ad0

Let me give more information about my case.  there is no X server running. I plan to do offscreen rendering.  There are two Radeon Vega 64 cards on my machine,  currently, I pick the first physical GPU card to create Vulkan logical device.

amingriyue, 2020-03-19:
Can you build a debug amdvlk.so with -DCMAKE_BUILD_TYPE=Debug option by following README.md?

Even you can simplify a simple test program, I can verify it locally.

renguannan1, 2020-03-19:
Info updated.

After I compiled AMDVLK source code with -DCMAKE_BUILD_TYPE=Debug. I met a new Segfault.
bt
(gdb) bt
#0  __GI_raise (sig=<optimized out>) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007fffeed0a166 in vk::VerifyExtensions (dev=...) at /root/vulkandriver/drivers/xgl/icd/api/vk_physical_device.cpp:6005
#2  0x00007fffeed0a29b in vk::VerifyProperties (device=...) at /root/vulkandriver/drivers/xgl/icd/api/vk_physical_device.cpp:6048
#3  0x00007fffeecf986c in vk::PhysicalDevice::LateInitialize (this=0x555555c6a8a8) at /root/vulkandriver/drivers/xgl/icd/api/vk_physical_device.cpp:1063
#4  0x00007fffeed28deb in vk::Instance::Init (this=0x555555af62e0, pAppInfo=0x7fffffffdff0) at /root/vulkandriver/drivers/xgl/icd/api/vk_instance.cpp:448
#5  0x00007fffeed28786 in vk::Instance::Create (pCreateInfo=0x7fffffffda90, pAllocator=0x0, pInstance=0x555555af5828) at /root/vulkandriver/drivers/xgl/icd/api/vk_instance.cpp:238
#6  0x00007fffeed2b215 in vk::entry::vkCreateInstance (pCreateInfo=0x7fffffffda90, pAllocator=0x0, pInstance=0x555555af5828) at /root/vulkandriver/drivers/xgl/icd/api/vk_instance.cpp:1393

And I tried to fix it. I simplified doing the following work. It worked about the new segfault.
Please help check if it is known bug or occurrs only in my environment. 
Anyway, If it only happens in my env,  I would better find out the root cause.

diff --git a/icd/api/vk_physical_device.cpp b/icd/api/vk_physical_device.cpp
index 20abcf6..4220478 100644
--- a/icd/api/vk_physical_device.cpp
+++ b/icd/api/vk_physical_device.cpp
@@ -6024,7 +6024,7 @@ static void VerifyExtensions(
                && dev.IsExtensionSupported(DeviceExtensions::KHR_SHADER_FLOAT16_INT8)
                && dev.IsExtensionSupported(DeviceExtensions::EXT_SHADER_VIEWPORT_INDEX_LAYER)
                && dev.IsExtensionSupported(DeviceExtensions::KHR_SHADER_SUBGROUP_EXTENDED_TYPES)
-               && dev.IsExtensionSupported(DeviceExtensions::KHR_TIMELINE_SEMAPHORE)
+//               && dev.IsExtensionSupported(DeviceExtensions::KHR_TIMELINE_SEMAPHORE)
                && dev.IsExtensionSupported(DeviceExtensions::KHR_UNIFORM_BUFFER_STANDARD_LAYOUT)
                && dev.IsExtensionSupported(DeviceExtensions::KHR_VULKAN_MEMORY_MODEL));

amingriyue, 2020-03-20:
Is the sentence you commented out is under #if VKI_SDK_1_2?

Vulkan1.2 will require timeline layer enablement.

Two resolve methods:

1. if you don't use timeline in your app, you can just comment it out.

2. Normal way is you install timeline layer manually, https://github.com/KhronosGroup/Vulkan-ExtensionLayer
steps are:
a. download it to your local, make and make install
b. export VK_INSTANCE_LAYERS="VK_LAYER_KHRONOS_timeline_semaphore"
export VK_LAYER_PATH="/usr/local/share/vulkan/explicit_layer.d"
export LD_LIBRARY_PATH="/usr/local/lib" 

Sorry for that, we should add this into Readme.

jinjianrong, 2020-03-20:
This is mentioned in the readme: https://github.com/GPUOpen-Drivers/AMDVLK#known-issues

renguannan1, 2020-03-20:
Thanks I installed the VK_LAYER_KHRONOS_timeline_semaphore layer.
But the root cause is probably the missing of CAP of AMD radeon vega64 card.
/dev/dri/renderD128 is the target card.
the checking code is as follows:
...
        int32_t fd = open("/dev/dri/renderD128", O_RDWR, 0);
        result = drmGetCap(fd, DRM_CAP_SYNCOBJ_TIMELINE, &cap);
        if (result < 0) {
                printf("drmGetCap error\n");
                goto out;
        }
...
It reports error and seems no drmGetCap capability.

amdvlk-admin, 2020-03-20:
That is expected if the kernel is older than 5.5, https://github.com/GPUOpen-Drivers/AMDVLK#known-issues .

renguannan1, 2020-03-20:
Okay. I am using Linux ubuntu 4.15.0-91-generic offical distro release.
Could you guys please recommand the right AMDVLK version to me ? the software and hardware INFO is described in my first post.  PS: I actually met several bugs in v-2020.Q1.3.

renguannan1, 2020-03-20:
The AMD offical websit give drivers on ubuntu 18.04 is amdgpu-pro-19.50-967956-ubuntu-18.04.tar.xz
its URL: https://www.amd.com/en/support/graphics/radeon-rx-vega-series/radeon-rx-vega-series/radeon-rx-vega-64
I used it in the first place, unfortunately, I met the vkBindImageMemory2() Segfault.
then I changed to newer upstream version then more bugs cames out. Any further ideas?
BTW, Could you tell me the upstream release tag corresponding to amdgpu-pro-19.50-967956-ubuntu-18.04.tar.xz, so I can dig into the code.

amingriyue, 2020-03-20:
As soon as you use SDK1.2,  you cannot avoid timeline extension.
1. After you comment out, Can the debug driver work for you? Is there still segfault in vkBindImageMemory2()?
2. After you installed the timeline layer, Is there any further issue?

renguannan1, 2020-03-20:
After commenting out the code line 
"dev.IsExtensionSupported(DeviceExtensions::KHR_TIMELINE_SEMAPHORE)" A NEW big Segfault occures when calling vkCreateDevice() Vulkan API.

#1  0x00007ffff06b1299 in Pal::Amdgpu::GpuMemory::AllocateOrPinMemory (this=0x555555d603f0, baseVirtAddr=4295626752, pPagingFence=0x555555ae4798, 
    virtualAccessMode=Pal::VirtualGpuMemAccessMode::Undefined, multiDeviceGpuMemoryCount=0, ppDevice=0x0, ppImage=0x0)
    at /root/vulkandriver/AMDVLK-v-2020.Q1.3/amdvlk_src/pal/src/core/os/amdgpu/amdgpuGpuMemory.cpp:260
#2  0x00007ffff063eb3f in Pal::GpuMemory::Init (this=0x555555d603f0, createInfo=..., internalInfo=...) at /root/vulkandriver/AMDVLK-v-2020.Q1.3/amdvlk_src/pal/src/core/gpuMemory.cpp:616
#3  0x00007ffff06b1625 in Pal::Amdgpu::GpuMemory::Init (this=0x555555d603f0, createInfo=..., internalInfo=...)
    at /root/vulkandriver/AMDVLK-v-2020.Q1.3/amdvlk_src/pal/src/core/os/amdgpu/amdgpuGpuMemory.cpp:351
...
#25 0x00007fffeec1f129 in vk::Device::Create (pPhysicalDevice=0x555555ad4c48, pCreateInfo=0x7fffffffd8e0, pAllocator=0x5555559611b8, ppDevice=0x5555557b3f30)
    at /root/vulkandriver/AMDVLK-v-2020.Q1.3/amdvlk_src/xgl/icd/api/vk_device.cpp:961
#26 0x00007fffeec42a44 in vk::PhysicalDevice::CreateDevice (this=0x555555ad4c48, pCreateInfo=0x7fffffffd8e0, pAllocator=0x5555559611b8, pDevice=0x5555557b3f30)
    at /root/vulkandriver/AMDVLK-v-2020.Q1.3/amdvlk_src/xgl/icd/api/vk_physical_device.cpp:1096
#27 0x00007fffeec54c21 in vk::entry::vkCreateDevice (physicalDevice=0x555555ad4c40, pCreateInfo=0x7fffffffd8e0, pAllocator=0x0, pDevice=0x5555557b3f30)
    at /root/vulkandriver/AMDVLK-v-2020.Q1.3/amdvlk_src/xgl/icd/api/vk_physical_device.cpp:6408

amingriyue, 2020-03-20:
That's impossible, it must have something missed.
I will find explicit steps to setup amdvlk with vulkan1.2 from internal team.

amingriyue, 2020-03-20:
Hi,
Our debug driver often have noise assert, Could you add below settings to /etc/amd/amdPalSettings.cfg:
SoftAssert,0
HardAssert,0

renguannan1, 2020-03-20:
thanks~, With this config file setting.  It finally reaches the vkBindImageMemory2() segfault.
NO any AMDVLK backtrace printing out. but with PAL error as follows:
AMD-PAL: Warn: Unconditional Alert | Reason: Unknown (/root/vulkandriver/AMDVLK-v-2020.Q1.3/amdvlk_src/pal/src/core/os/amdgpu/amdgpuGpuMemory.cpp:257:AllocateOrPinMemory)
AMD-PAL: Error: Code Not Tested! | Reason: Unknown (/root/vulkandriver/AMDVLK-v-2020.Q1.3/amdvlk_src/pal/src/core/os/amdgpu/amdgpuGpuMemory.cpp:260:AllocateOrPinMemory)
does it seem that the AMD-PAL error leads to vkBindImageMemory2() segfault ?

amingriyue, 2020-03-20:
》》AMD-PAL: Error: Code Not Tested! | Reason: Unknown (/root/vulkandriver/AMDVLK-v-2020.Q1.3/amdvlk_src/pal/src/core/os/amdgpu/amdgpuGpuMemory.cpp:260:AllocateOrPinMemory)

My teammate points the reason for this info is your card isn't setup correctly.

Can you show your dmesg?

renguannan1, 2020-03-20:
I cannnot find where to attach dmesg log file. I sent you a email. Please help performa analysis.
if any info in the log is related to the issue, you can paste it out, let other guys follow easily.

amingriyue, 2020-03-20:
I've checked your dmesg, it's normal.
Can you execute "sudo cat /sys/kernel/debug/dri/0/amdgpu_vram_mm"?

renguannan1, 2020-03-20:
0x0000000000000000-0x0000000000000900: 2304: used
0x0000000000000900-0x0000000000000a00: 256: used
0x0000000000000a00-0x0000000000000a01: 1: used
0x0000000000000a01-0x0000000000000a02: 1: used
0x0000000000000a02-0x0000000000000a0a: 8: used
0x0000000000000a0a-0x0000000000000a0b: 1: used
0x0000000000000a0b-0x0000000000000a0c: 1: used
0x0000000000000a0c-0x0000000000000a0d: 1: used
0x0000000000000a0d-0x0000000000000a0e: 1: used
0x0000000000000a0e-0x0000000000000a27: 25: used
0x0000000000000a27-0x0000000000000a28: 1: used
0x0000000000000a28-0x0000000000000c8e: 614: used
0x0000000000000c8e-0x0000000000000ceb: 93: used
0x0000000000000ceb-0x0000000000000cec: 1: used
0x0000000000000cec-0x0000000000000ced: 1: used
0x0000000000000ced-0x0000000000000cee: 1: used
0x0000000000000cee-0x0000000000000cf2: 4: used
0x0000000000000cf2-0x0000000000000d32: 64: used
0x0000000000000d32-0x0000000000000d72: 64: used
0x0000000000000d72-0x0000000000001000: 654: free
0x0000000000001000-0x0000000000001400: 1024: used
0x0000000000001400-0x00000000001fefe9: 2087913: free
0x00000000001fefe9-0x00000000001fefea: 1: used
0x00000000001fefea-0x00000000001fefeb: 1: used
0x00000000001fefeb-0x00000000001fefec: 1: used
0x00000000001fefec-0x00000000001fefed: 1: used
0x00000000001fefed-0x00000000001fefee: 1: used
0x00000000001fefee-0x00000000001fefef: 1: used
0x00000000001fefef-0x00000000001feff0: 1: used
0x00000000001feff0-0x00000000001ff000: 16: used
total: 2093056, used 4489 free 2088567
man size:2093056 pages, ram usage:17MB, vis usage:17MB

amingriyue, 2020-03-20:
you are hitting here in GpuMemory::AllocateOrPinMemory
                    if (validHeapFound == false)
                    {
                        // Provide some info that we're getting into this path
                        PAL_ALERT_ALWAYS();
                        // Duplication of Windows path
                        PAL_NOT_TESTED();

Can you set a break point in /root/vulkandriver/AMDVLK-v-2020.Q1.3/amdvlk_src/pal/src/core/os/amdgpu/amdgpuGpuMemory.cpp:254 ?
   >>>>                if (validHeapFound == false)
when driver stops here, you execute "sudo cat /sys/kernel/debug/dri/0/amdgpu_vram_mm".

renguannan1, 2020-03-20:
root@ubuntu:~/vulkandriver/AMDVLK-v-2020.Q1.3# cat /sys/kernel/debug/dri/0/amdgpu_vram_mm
0x0000000000000000-0x0000000000000900: 2304: used
0x0000000000000900-0x0000000000000a00: 256: used
0x0000000000000a00-0x0000000000000a01: 1: used
0x0000000000000a01-0x0000000000000a02: 1: used
0x0000000000000a02-0x0000000000000a0a: 8: used
0x0000000000000a0a-0x0000000000000a0b: 1: used
0x0000000000000a0b-0x0000000000000a0c: 1: used
0x0000000000000a0c-0x0000000000000a0d: 1: used
0x0000000000000a0d-0x0000000000000a0e: 1: used
0x0000000000000a0e-0x0000000000000a27: 25: used
0x0000000000000a27-0x0000000000000a28: 1: used
0x0000000000000a28-0x0000000000000c8e: 614: used
0x0000000000000c8e-0x0000000000000ceb: 93: used
0x0000000000000ceb-0x0000000000000cec: 1: used
0x0000000000000cec-0x0000000000000ced: 1: used
0x0000000000000ced-0x0000000000000cee: 1: used
0x0000000000000cee-0x0000000000000cf2: 4: used
0x0000000000000cf2-0x0000000000001000: 782: free
0x0000000000001000-0x0000000000001400: 1024: used
0x0000000000001400-0x00000000001fefec: 2087916: free
0x00000000001fefec-0x00000000001fefed: 1: used
0x00000000001fefed-0x00000000001fefee: 1: used
0x00000000001fefee-0x00000000001feff0: 2: free
0x00000000001feff0-0x00000000001ff000: 16: used
total: 2093056, used 4356 free 2088700
man size:2093056 pages, ram usage:17MB, vis usage:17MB

amingriyue, 2020-03-20:
second cat isn't sense to me. Because when you stop at break point line 254, that means no vram can be used.
You'd better to simplify a testing program to us, we can locally verify it.

Another method to identify the problem is from your program or amdvlk driver, you can cts with below cases:
dEQP-VK.memory.binding*

If above all passed, that means vkBindImageMemory2() works fine.

renguannan1, 2020-03-20:
I send the test codes to you just now, pls have a check. Thanks

renguannan1, 2020-03-20:
I tried CTS test suits. It reports the same errors

AMD-PAL: Warn: Unconditional Alert | Reason: Unknown (/root/vulkandriver/AMDVLK-v-2020.Q1.3/amdvlk_src/pal/src/core/os/amdgpu/amdgpuGpuMemory.cpp:257:AllocateOrPinMemory)^M
AMD-PAL: Error: Code Not Tested! | Reason: Unknown (/root/vulkandriver/AMDVLK-v-2020.Q1.3/amdvlk_src/pal/src/core/os/amdgpu/amdgpuGpuMemory.cpp:260:AllocateOrPinMemory)^M
AMD-PAL: Warn: Unconditional Alert | Reason: Failed to get function pointer for: amdgpu_cs_syncobj_query2 (/root/vulkandriver/AMDVLK-v-2020.Q1.3/amdvlk_src/pal/inc/util/palLibrary.h:84:GetFunction)

amingriyue, 2020-03-20:
Can you enter vkBindImageMemory2 in amdvlk icd driver?

I tried CTS, all dEQP-VK.memory.binding* cases are passed.
But I reproduce this with your program, the segfault happens in vulkan loader, vulkan loader failed to redirect vkBindImageMemory2 to icd driver. I guess somewhere entry table is established incurrectly.

amingriyue, 2020-03-20:
Finally, locate the root cause, vkBindImageMemory2 requires API version >= 1.1
So you should change  appInfo.apiVersion to 1.1 as below:

        VkApplicationInfo appInfo = {};
        appInfo.sType = VK_STRUCTURE_TYPE_APPLICATION_INFO;
        appInfo.pApplicationName = "AMDGPU";
        appInfo.applicationVersion = VK_MAKE_VERSION(1, 0, 0);
        appInfo.pEngineName = "No Engine";
        appInfo.engineVersion = VK_MAKE_VERSION(1, 0, 0);
        appInfo.apiVersion = VK_API_VERSION_1_1;

renguannan1, 2020-03-20:
Thank you, it is very helpful!
To be honest.  vulkan API is not strong enough in terms of APP writing.
BTW, please help look into issue and make sure it is okay 
the  AMD-PAL error amdgpuGpuMemory.cpp:260:AllocateOrPinMemory

renguannan1, 2020-03-20:
I think It's okay to close the issue.  The other issues we met above can be followd by opening a  new issue.

closed 2020-03-22 by jinjianrong

amingriyue, 2020-03-23:
Maybe amdvlk should improve the checking of extension for different version.

>the AMD-PAL error amdgpuGpuMemory.cpp:260:AllocateOrPinMemory
This is good info to mention you run into incorrect path, I suspect it's most possible that you application pass wrong heap for memory allocation.