GPUOpen-Drivers/AMDVLK #147: build failed with makefile2 error

ggghamd, 2020-03-23:
Followed the instruction in README in AMDVKL folder for ubuntu64 build, but build fails. Either instruction is incomplete, or build issue in the branch. 
 2085  cmake -H. -Bbuilds/Release64
 2086  git branch
 2087  ls
 2088  cat built.log
 2089  cd cd builds/Release64
 2090  make -j$(nproc)
 2091  cd cd builds/Release64
 2092  make -j$(nproc)

[ 27%] Building CXX object llvm/utils/TableGen/CMakeFiles/obj.llvm-tblgen.dir/X86ModRMFilters.cpp.o
[ 27%] Building CXX object llvm/utils/TableGen/CMakeFiles/obj.llvm-tblgen.dir/X86RecognizableInstr.cpp.o
[ 27%] Building CXX object llvm/utils/TableGen/CMakeFiles/obj.llvm-tblgen.dir/WebAssemblyDisassemblerEmitter.cpp.o
[ 27%] Building CXX object llvm/utils/TableGen/CMakeFiles/obj.llvm-tblgen.dir/CTagsEmitter.cpp.o
[ 27%] Linking CXX static library ../libLLVMTextAPI.a
[ 27%] Built target LLVMTextAPI
[ 27%] Linking CXX static library libvfx.a
[ 27%] Built target vfx
[ 27%] Linking CXX static library ../libLLVMObjectYAML.a
[ 27%] Built target LLVMObjectYAML
[ 27%] Linking CXX static library ../../libLLVMMCParser.a
[ 27%] Linking CXX static library ../libLLVMSupport.a
[ 27%] Built target LLVMMCParser
[ 27%] Built target LLVMSupport
[ 27%] Linking CXX static library ../../libLLVMDebugInfoPDB.a
[ 27%] Built target LLVMDebugInfoPDB

[ 27%] Linking CXX static library ../../libLLVMDebugInfoDWARF.a
[ 27%] Built target LLVMDebugInfoDWARF
[ 28%] Linking CXX static library ../libLLVMMC.a
[ 28%] Built target LLVMMC
[ 28%] Linking CXX static library ../../libLLVMDebugInfoCodeView.a
[ 28%] Built target LLVMDebugInfoCodeView
[ 28%] Built target obj.llvm-tblgen
make: *** [Makefile:152: all] Error 2

jinjianrong, 2020-03-23:
Where do you get the llvm code? is it from https://github.com/GPUOpen-Drivers/llvm-project?

ggghamd, 2020-04-07:
I followed this instruction: https://github.com/RadeonOpenCompute/ROCm/tree/roc-3.1.0#Getting-the-ROCm-Source-Code
I tried on 3.3 but instruction were no longer there and could not download the same way as in 3.1

ggghamd, 2020-04-09:
Hi, it loosk like I mixed up. I was able to build amdvlk after switchin to stable branch. I think this can close.

closed 2020-07-16 by JacobHeAMD