GPUOpen-Drivers/AMDVLK #148: build requirement failed

ggghamd, 2020-03-23:
root@sriov-guest:/git.co/gpuopen/AMDVLK# lsb_release --all ; uname -r
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.4 LTS
Release:        18.04
Codename:       bionic
5.3.0-42-generic
root@sriov-guest:/git.c

root@sriov-guest:/git.co/gpuopen/AMDVLK# cat  README.md  | grep "sudo apt"
sudo apt-get install build-essential python3 cmake curl g++-multilib gcc-multilib
sudo apt-get install libssl-dev libssl-dev:i386 libx11-dev libxcb1-dev x11proto-dri2-dev libxcb-dri3-dev libxcb-dri2-0-dev libxcb-present-dev libxshmfence-dev libxrandr-dev libx11-dev:i386 libxcb1-dev:i386 x11proto-dri2-dev:i386 libxcb-dri3-dev:i386 libxcb-dri2-0-dev:i386 libxcb-present-dev:i386 libxshmfence-dev:i386 libwayland-dev libwayland-dev:i386 libxrandr-dev:i386

root@sriov-guest:/git.co/gpuopen/AMDVLK# sudo apt-get install libssl-dev libssl-dev:i386 libx11-dev libxcb1-dev x11proto-dri2-dev libxcb-dri3-dev libxcb-dri2-0-dev libxcb-present-dev libxshmfence-dev libxrandr-dev libx11-dev:i386 libxcb1-dev:i386 x11proto-dri2-dev:i386 libxcb-dri3-dev:i386 libxcb-dri2-0-dev:i386 libxcb-present-dev:i386 libxshmfence-dev:i386 libwayland-dev libwayland-dev:i386 libxrandr-dev:i386
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package x11proto-dri2-dev:i386 is a virtual package provided by:
  x11proto-dri2-dev 2018.4-4
  x11proto-dev 2018.4-4
You should explicitly select one to install.

JaxLinAMD, 2020-03-23:
I have submitted the fix, so close this ticket.

closed 2020-03-23 by JaxLinAMD

ggghamd, 2020-04-07:
i think fix is adequate