GPUOpen-Drivers/AMDVLK #150: VK_KHR_ray_tracing support?

SamHSmith, 2020-03-25:
This might sound crazy considering it was only released a couple days ago but. When can we expect software VK_KHR_ray_tracing implementation for navi etc?

oscarbg, 2020-05-09:
also with current news that RDNA2 might have DX12 Ultimate support also curious how AMDVLK will expose mesh shaders , variable shading and sampler feedback on RDNA2.. as currently these features are exposed on Vulkan as NV extensions only..
hope AMD is working with Khronos on exposing EXT or KHR variants of:
*VK_NV_mesh_shader
*VK_NV_shading_rate_image
*VK_NV_shader_image_footprint & VK_NV_representative_fragment_test 

at least for variable shading we have an EXT variant already:
VK_EXT_fragment_density_map 
altough there are diffs..

JaxLinAMD, 2020-07-24:
I am also wondering when VK_KHR_ray_tracing implementation can be released :)

closed 2020-07-24 by JaxLinAMD