GPUOpen-Drivers/AMDVLK #151: [2020.Q1.3] Error adding symbols:DSO missing from command line

RarogCmex, 2020-03-25:
/usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: /var/tmp/portage/media-libs/amdvlk-2020.1.3/work/build-abi_x86_32.x86/llvm/lib/libLLVMAMDGPUDesc.so.11git: error adding symbols: DSO missing from command line.
All source code are fetched properly.
[build.log](https://github.com/GPUOpen-Drivers/AMDVLK/files/4383929/build.log)
Operating system data:
[os.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/4383932/os.txt)

JaxLinAMD, 2020-03-26:
Hi, @RarogCmex , could you try if following change can fix the build failure?
in xgl/CMakeLists.txt, add following lines right after project(XGL VERSION 1 LANGUAGES C CXX):

if(UNIX)
    if("${CMAKE_CXX_COMPILER_ID}" STREQUAL "GNU")
        set(CMAKE_AR /usr/bin/gcc-ar)
        set(CMAKE_RANLIB /usr/bin/gcc-ranlib)
    endif()
endif()

thanks

RarogCmex, 2020-03-26:
Thank you so for fast answer, but it could not help:
[build.log](https://github.com/GPUOpen-Drivers/AMDVLK/files/4386393/build.log)

RarogCmex, 2020-03-26:
The source code is downloading with following:
XGL_COMMIT="80e5a4b11ad2058097e77746772ddc9ab2118e07"
PAL_COMMIT="e642f608a62887d40d1f25509d2951a4a3576985"
LLPC_COMMIT="fc21c950b629753f9cc7d0941937c57262ceadcf"
SPVGEN_COMMIT="843c6b95f731589bf497fad29dadf7fda4934aad"
LLVM_PROJECT_COMMIT="e404e4b2db325184dbc2d14f31ef891d938f3835"
METROHASH_COMMIT="2b6fee002db6cc92345b02aeee963ebaaf4c0e2f"
CWPACK_COMMIT="b601c88aeca7a7b08becb3d32709de383c8ee428"
## SRC_URI
SRC_URI=" ${FETCH_URI}/AMDVLK/archive/${CORRECT_AMDVLK_PV}.tar.gz -> AMDVLK-${CORRECT_AMDVLK_PV}.tar.gz
${FETCH_URI}/xgl/archive/${XGL_COMMIT}.tar.gz -> xgl-${XGL_COMMIT}.tar.gz
${FETCH_URI}/pal/archive/${PAL_COMMIT}.tar.gz -> pal-${PAL_COMMIT}.tar.gz
${FETCH_URI}/llpc/archive/${LLPC_COMMIT}.tar.gz -> llpc-${LLPC_COMMIT}.tar.gz
${FETCH_URI}/spvgen/archive/${SPVGEN_COMMIT}.tar.gz -> spvgen-${SPVGEN_COMMIT}.tar.gz
${FETCH_URI}/llvm-project/archive/${LLVM_PROJECT_COMMIT}.tar.gz -> llvm-project-${LLVM_PROJECT_COMMIT}.tar.gz
${FETCH_URI}/MetroHash/archive/${METROHASH_COMMIT}.tar.gz -> MetroHash-${METROHASH_COMMIT}.tar.gz
${FETCH_URI}/CWPack/archive/${CWPACK_COMMIT}.tar.gz -> CWPack-${CWPACK_COMMIT}.tar.gz"

###EBUILD FUNCTIONS
src_prepare() {
##moving src to proper directories
mkdir -p ${S}; mkdir -p ${S}/third_party
mv AMDVLK-${CORRECT_AMDVLK_PV}/ ${S}/AMDVLK
mv xgl-${XGL_COMMIT}/ ${S}/xgl
mv pal-${PAL_COMMIT}/ ${S}/pal
mv llpc-${LLPC_COMMIT}/ ${S}/llpc
mv spvgen-${SPVGEN_COMMIT}/ ${S}/spvgen
mv llvm-project-${LLVM_PROJECT_COMMIT}/ ${S}/llvm-project
mv MetroHash-${METROHASH_COMMIT}/ ${S}/third_party/metrohash
mv CWPack-${CWPACK_COMMIT}/ ${S}/third_party/cwpack
cat << EOF > "${T}/10-amdvlk-dri3.conf" || die

RarogCmex, 2020-03-26:
Complete ebuild
[amdvlk-2020.1.3.ebuild.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/4388731/amdvlk-2020.1.3.ebuild.txt)

RarogCmex, 2020-03-26:
The following snippet of build.log:
FAILED: llpc/amdllpc

RarogCmex, 2020-03-27:
The situation is clarified. AMDVLK doesn't support full building with cmake, it seems that compiling with cmake broken. When I switch to more generic method of ebuild it become working. 
So I'm not sure that is issue.

JaxLinAMD, 2020-03-30:
@RarogCmex , what do you mean by "ebuild"? Did you use some special method to build driver so that driver build is successful?

Johnnynator, 2020-03-30:
@RarogCmex Can you provide the full cmake command and env vars in use when compiling?

> @RarogCmex , what do you mean by "ebuild"? Did you use some special method to build driver so that driver build is successful?

Ebuild is the Gentoo name for a build definition for a package.

RarogCmex, 2020-03-30:
I can't do it in this moment, sorry me.

JaxLinAMD, 2020-04-09:
I need to setup Gentoo enviroment to check what's happening

RarogCmex, 2020-04-09:
@JaxLinAMD Ok, thanks, here is how to install Gentoo https://wiki.gentoo.org/wiki/Handbook:AMD64/Full
with addition that instead building kernel in step 3 you may get more experimental ready-to-use ones: sys-kernel/gentoo-kernel
Mail me denis7774@gmail.com if you're experiencing issues.
I'll send ebuilds to test with it within a day.

RarogCmex, 2020-04-09:
@JaxLinAMD 
Also instead of full installation you may just chroot.
Here is ebuild (rename: erase all after the amdvlk-2020.1.4.ebuild )
[amdvlk-2020.1.4.ebuild.works.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/4456049/amdvlk-2020.1.4.ebuild.works.txt)

[amdvlk-2020.1.4.ebuild.not_works.txt](https://github.com/GPUOpen-Drivers/AMDVLK/files/4456048/amdvlk-2020.1.4.ebuild.not_works.txt)

You may just put it to /var/db/repos/gentoo (but make sure that there is no overwrites) or just create new ebuild repository https://wiki.gentoo.org/wiki/Ebuild_repository

JaxLinAMD, 2020-04-15:
@RarogCmex , I found that driver is build through cmake utils provided by ebuild, which will generate build error, have you tried to build driver directly using cmake( no ebuild, only use cmake and make):

cmake -H. ***
make

RarogCmex, 2020-04-15:
@JaxLinAMD Yes, see amdvlk-2020.1.4.ebuild.works.txt

JaxLinAMD, 2020-04-20:
@RarogCmex , could you try to build with cmake directly in console without the help of ebuild to see if there is any error?
cmake -H. -Bbuilds/Release64
make -j8
I can compile successfully with cmake, in fact, I am not quite familiar how use ebuild.

RarogCmex, 2020-04-20:
@JaxLinAMD Yes, It works without errors.
Maybe it's portage cmake build system error, I don't have heath at this time
P.S. So do I about ebuilds, but I tries :)

closed 2020-07-24 by JaxLinAMD