GPUOpen-Drivers/AMDVLK #152: [2020.Q1.4] potencial issue: array subscript 1 is outside array bounds of ‘int [1]

RarogCmex, 2020-03-27:
Gentoo portage build system found a potential issue 
* QA Notice: Package triggers severe warnings which indicate that it
 *            may exhibit random runtime failures.
 * /var/tmp/portage/media-libs/amdvlk-2020.1.4/work/llvm-project/llvm/include/llvm/ADT/SmallVector.h:520:7: warning: array subscript 1 is outside array bounds of ‘int [1]’ [-Warray-bounds]

dstutt, 2020-03-30:
Thanks for the info.

Have you got any more context for the warning? It's not SmallVector itself that's the problem I think.

RarogCmex, 2020-03-30:
I can't do it in this moment, sorry me.

kuhar, 2020-04-07:
I think I'm getting the same warning:
```
[697/2253] Building CXX object llvm/lib/CodeGen/CMakeFiles/LLVMCodeGen.dir/ExecutionDomainFix.cpp.o
In file included from /ssd/opensource/amdvlk/vulkandriver/drivers/llvm-project/llvm/include/llvm/CodeGen/ExecutionDomainFix.h:25,
                 from /ssd/opensource/amdvlk/vulkandriver/drivers/llvm-project/llvm/lib/CodeGen/ExecutionDomainFix.cpp:9:
/ssd/opensource/amdvlk/vulkandriver/drivers/llvm-project/llvm/include/llvm/ADT/SmallVector.h: In member function ‘void llvm::ExecutionDomainFix::visitSoftInstr(llvm::MachineInstr*, unsigned int)’:
/ssd/opensource/amdvlk/vulkandriver/drivers/llvm-project/llvm/include/llvm/ADT/SmallVector.h:520:7: warning: array subscript 1 is outside array bounds of ‘int [1]’ [-Warray-bounds]
  520 |       ++EltPtr;
      |       ^~
/ssd/opensource/amdvlk/vulkandriver/drivers/llvm-project/llvm/lib/CodeGen/ExecutionDomainFix.cpp:330:12: note: while referencing ‘rx’
  330 |   for (int rx : used) {
      | 
```

This is on the dev branch from today (April 7 2020) with the following cmake invocation:
```
cmake -H. -Bbuilds/Release64 -GNinja
```

```
$ gcc --version                                                                   
gcc (Debian 9.2.1-19) 9.2.1 20191109
```
Seems like a false positive to me.

dstutt, 2020-04-15:
Yes, false positive.

closed 2020-04-16 by jinjianrong