GPUOpen-Drivers/AMDVLK #153: Please add hint regarding udev rule to README.md to help people getting this driver to work on other distros than Ubuntu and RHEL

DiabeticCrab, 2020-03-28:
Before adding this udev rule, I couldn't run amdvlk on Gentoo x64 w/ Vega 56, libdrm 2.4.100 and Kernel 5.5.10. The debug build of amdvlk was complaining about not getting function pointers to libdrm related stuff. Now they work flawlessly, the self compiled version, aswell as the ones (regular and pro) ripped out of the Ubuntu deb.

> /etc/udev/rules.d/91-amdvlk-modeset.rules
`KERNEL=="controlD[0-9]*", SUBSYSTEM=="drm", MODE="0600"`

JaxLinAMD, 2020-03-30:
Hi @DiabeticCrab , I'm confused why this rule is needed, could you find /dev/dri/controlD* on your system? there should be only card0 and renderD128. thanks.

JaxLinAMD, 2020-03-30:
by the way, 91-amdvlk-modeset.rules, is this file created by you manually?

DiabeticCrab, 2020-03-30:
I don't have the /dev/dri/controlD* device files on my system, so I don't know why the udev rule helped. All I know is that I installed it, reloaded the udev rules and suddenly the drivers were working.
 
The file was taken from the Ubuntu AMDVLK-Pro 19.50 deb (libdrm2-amdgpu_2.4.99-967956_amd64.deb/data.tar.xz/lib/udev/rules.d/91-amdgpu-pro-modeset.rules) in an attempt to get things working.

I tried reverting my system stepwise (removed the udev rule, downgraded some packages) to a state where the error still occurs, in order to get behind the exact cause but to no avail (drv still works), so I am stuck now. I did not recompile/update the kernel, so I can rule that out at least.

Here is a log of my recently installed, removed and updated packages from the last 3 days. Maybe you can see a clue like a required or critical package being installed/updated. 
I think the drivers started working on Saturday: [genlop.log](https://github.com/GPUOpen-Drivers/AMDVLK/files/4403424/genlop.log)

JaxLinAMD, 2020-03-31:
I checked the log but  couldn't find any useful information form genlog.log.

"The debug build of amdvlk was complaining about not getting function pointers to libdrm related stuff"
 -- could you provide the log amdvlk printed? can release build of amdvlk work?

JaxLinAMD, 2020-03-31:
could you also provide libdrm version on your system?

JaxLinAMD, 2020-04-09:
I need to setup Gentoo environment to check what's the problem.

Flakebi, 2020-04-09:
Hm, I think I might be hitting the same issue. I’m not completely sure, @DiabeticCrab can you post your error message too?
A build with assertions enabled throws an assertion in [`vk_physical_device.cpp:VerifyExtensions`](https://github.com/GPUOpen-Drivers/xgl/blob/ffd39443fec0ce821b7702173eba7184855c7b5e/icd/api/vk_physical_device.cpp#L6027) because `KHR_TIMELINE_SEMAPHORE` is not available.
If I understand the pal and libdrm source code right, this extension needs at least [libdrm 2.4.101](https://lwn.net/Articles/816779/) for the 'syncobj query interface v2'.
Otherwise, `KHR_TIMELINE_SEMAPHORE` cannot be enabled because of a missing function in libdrm. Xgl assumes that we support Vulkan 1.2, which needs `KHR_TIMELINE_SEMAPHORE`, so the assertion fails.

So this should be reproducible on other OSes too.

JaxLinAMD, 2020-04-09:
Ah, yes, I just remembered that amdvlk debug driver can't work.

JaxLinAMD, 2020-04-09:
@DiabeticCrab, Could you provide the error log ? I just remembered that amdvlk debug driver can't work because of KHR_TIMELINE_SEMAPHORE is not supported.

Flakebi, 2020-04-09:
We should fix this but I’m not sure how. I think we shouldn’t expose that we support Vulkan 1.2 in the case that libdrm is to old?

DiabeticCrab, 2020-04-09:
Sorry for not replying for so long, but I was trying to recreate the issue and I finally found it. Turns out, my linux-headers (4.19) were mismatched with the installed kernel (5.5). libdrm used is still 2.4.100 (0190f49a139e7069d7cad6a6890832831da1aa8b)
This is what happens in that case: 
[amdvlk64_dbg_vulkaninfo.log](https://github.com/GPUOpen-Drivers/AMDVLK/files/4456642/amdvlk64_dbg_vulkaninfo.log)

This might affect other distros with mismatched header versions aswell (i.e. Arch tends to have them a few versions behind iirc) or users with a custom kernel. The only set of headers that worked for me, were the manually installed ones from the kernel source tree (sudo make INSTALL_HDR_PATH=/usr headers_install). Using the 5.5 headers from the package manager did not work to my surprise.

JaxLinAMD, 2020-04-10:
Hi @DiabeticCrab, according to the log you provided, this is a known issue, we have tracked this issue internally, debug driver will trigger assertion for now. I have changed the issue title since the old tile is kind of misleading. We will make debug driver work as soon as possible.

JaxLinAMD, 2020-04-10:
building the driver with "-march=znver2" or "-march=native" leads to a segfault in applications

-- please use compile options recommended in ReadMe. We can't handle all the various compile options.
If you really want to create a issue for this, you'd better make it clear what "-march=znver2" or "-march=native" is for and make the topic more clear, don't create a issue with topic like " build failure with "-march=znver2" or "-march=native" ", use topic like "build failure for some specific architecture besides x86_64"

Flakebi, 2020-05-08:
As a note: It is possible to use a debug driver by adding `HardAssert,0` to `.config/amdPalSettings.cfg`, which makes the assertions non-fatal.

closed 2020-08-19 by JacobHeAMD