GPUOpen-Drivers/AMDVLK #159: make wants to link /usr/lib/x86_64-linux-gnu/libxml2.so when building 32-bit version in 64-bit environment

megabyte0, 2020-04-27:
```
/usr/lib/x86_64-linux-gnu/libxml2.so: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status
llvm/tools/llvm-mt/CMakeFiles/llvm-mt.dir/build.make:100: recipe for target 'llvm/bin/llvm-mt' failed
make[2]: *** [llvm/bin/llvm-mt] Error 1
CMakeFiles/Makefile2:17571: recipe for target 'llvm/tools/llvm-mt/CMakeFiles/llvm-mt.dir/all' failed
make[1]: *** [llvm/tools/llvm-mt/CMakeFiles/llvm-mt.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.4 LTS
Release:	18.04
Codename:	bionic
$ uname -a
Linux topagent8-Lenovo-ideapad-100-15IBD 4.15.0-96-generic #97-Ubuntu SMP Wed Apr 1 03:25:46 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
$ dpkg-query -S libxml2.so
libxml2:amd64: /usr/lib/x86_64-linux-gnu/libxml2.so.2.9.10
vagrant: /opt/vagrant/embedded/lib/libxml2.so
libxml2:amd64: /usr/lib/x86_64-linux-gnu/libxml2.so.2
vagrant: /opt/vagrant/embedded/lib/libxml2.so.2
libxml2:i386: /usr/lib/i386-linux-gnu/libxml2.so.2
libxml2-dev:amd64: /usr/lib/x86_64-linux-gnu/libxml2.so
libxml2-dev:i386: /usr/lib/i386-linux-gnu/libxml2.so
libxml2:i386: /usr/lib/i386-linux-gnu/libxml2.so.2.9.10
vagrant: /opt/vagrant/embedded/lib/libxml2.so.2.9.7
$ make --version
GNU Make 4.1
Built for x86_64-pc-linux-gnu
$ cmake --version
cmake version 3.10.2
$ ld --version
GNU ld (GNU Binutils for Ubuntu) 2.30
```

Flakebi, 2020-04-29:
I tried to reproduce this, but it works fine in a 64-bit docker container.
`Dockerfile`:
```Dockerfile
FROM ubuntu:18.04

RUN dpkg --add-architecture i386
RUN apt-get update
RUN apt-get install -yqq build-essential python3 cmake curl g++-multilib gcc-multilib libssl-dev libssl-dev:i386 libx11-dev libxcb1-dev x11proto-dri2-dev libxcb-dri3-dev libxcb-dri2-0-dev libxcb-present-dev libxshmfence-dev libxrandr-dev libx11-dev:i386 libxcb1-dev:i386 libxcb-dri3-dev:i386 libxcb-dri2-0-dev:i386 libxcb-present-dev:i386 libxshmfence-dev:i386 libwayland-dev libwayland-dev:i386 libxrandr-dev:i386 repo ninja-build pkg-config

RUN mkdir /driver_build
WORKDIR /driver_build
RUN repo init -u https://github.com/GPUOpen-Drivers/AMDVLK.git -b dev --no-repo-verify --depth=1 && repo sync
WORKDIR /driver_build/drivers/spvgen/external
RUN python2 fetch_external_sources.py

WORKDIR /driver_build/drivers/xgl
RUN mkdir Release64
WORKDIR /driver_build/drivers/xgl/Release64
RUN cmake -GNinja \
        -DCMAKE_BUILD_TYPE=Release \
        -DBUILD_WAYLAND_SUPPORT=ON \
        -DCMAKE_C_FLAGS=-m32 -DCMAKE_CXX_FLAGS=-m32 \
        ..

RUN ninja

CMD bash
```

```bash
$ docker build -t amdvlk32 .
…
$ docker run -it amdvlk32
root@47cf6ac9d44b:/driver_build/drivers/xgl/Release64# file icd/amdvlk32.so
icd/amdvlk32.so: ELF 32-bit LSB shared object, Intel 80386, version 1 (GNU/Linux), dynamically linked, BuildID[sha1]=ac56bd5fd8d083defd2a77e22cfbe0e1fdfdd3dc, stripped
```

megabyte0, 2020-04-29:
Sure it works fine, but try to add libxml2 libxml2:i386 libxml2-dev libxml2-dev:i386 to the apt-get install and see what happens. Or should I remove or somehow disable them before compiling? Thanks

Flakebi, 2020-04-30:
Imo it should still work even if additional packages are installed.
Installing both, `libxml2-dev` and `libxml2-dev:i386` fails for me because the dependencies `libicu-dev` and `libicu-dev:i386` are conflicting.
I checked both:
1. Installing `libxml2 libxml2:i386 libxml2-dev`: Running `cmake . && ninja` does nothing, so the x64 version of libxml2 is correctly not found
2. Installing `libxml2 libxml2:i386 libxml2-dev:i386`: Running `cmake . && ninja` finds libxml2, rebuilds a lot of stuff and succeeds in the end

So, I guess the problem is that you somehow managed to install both `libxml2-dev` and `libxml2-dev:i386` which are conflicting because `libicu-dev` and `libicu-dev:i386` probably contain files in the same path.
Can you try if removing all `libicu-dev` and `libxml2-dev` packages and reinstalling some of them fixes the problem?

megabyte0, 2020-05-01:
Thank you it worked. I uninstalled all the libicu-dev libicu-dev:i386 libxml2-dev libxml2-dev:i386 and installed only libicu-dev:i386 libxml2-dev:i386 then
It did find correctly now
```
-- Found LibXml2: /usr/lib/i386-linux-gnu/libxml2.so (found version "2.9.10")
```
and it's compiling now and went further behind the moment it did stop so I think it will succeed now. Thank you again)

Flakebi, 2020-05-04:
Glad you got it working.
I’m still baffled how you manage to install `libicu-dev` and `libicu-dev:i386` at the same time because these packages are conflicting on ubuntu 18.04 ([bug report](https://bugs.launchpad.net/ubuntu/+source/icu/+bug/1784135), both packages contain `icu-config` for [amd64](https://packages.ubuntu.com/bionic/amd64/libicu-dev/filelist) or [i386](https://packages.ubuntu.com/bionic/i386/libicu-dev/filelist) in the same path).
So I guess you have any extra repositories installed that change this?
<details>
  <summary><code>/etc/apt/sources.list</code> in docker looks like this</summary>

```bash
# See http://help.ubuntu.com/community/UpgradeNotes for how to upgrade to
# newer versions of the distribution.
deb http://archive.ubuntu.com/ubuntu/ bionic main restricted
# deb-src http://archive.ubuntu.com/ubuntu/ bionic main restricted

## Major bug fix updates produced after the final release of the
## distribution.
deb http://archive.ubuntu.com/ubuntu/ bionic-updates main restricted
# deb-src http://archive.ubuntu.com/ubuntu/ bionic-updates main restricted

## N.B. software from this repository is ENTIRELY UNSUPPORTED by the Ubuntu
## team. Also, please note that software in universe WILL NOT receive any
## review or updates from the Ubuntu security team.
deb http://archive.ubuntu.com/ubuntu/ bionic universe
# deb-src http://archive.ubuntu.com/ubuntu/ bionic universe
deb http://archive.ubuntu.com/ubuntu/ bionic-updates universe
# deb-src http://archive.ubuntu.com/ubuntu/ bionic-updates universe

## N.B. software from this repository is ENTIRELY UNSUPPORTED by the Ubuntu
## team, and may not be under a free licence. Please satisfy yourself as to
## your rights to use the software. Also, please note that software in
## multiverse WILL NOT receive any review or updates from the Ubuntu
## security team.
deb http://archive.ubuntu.com/ubuntu/ bionic multiverse
# deb-src http://archive.ubuntu.com/ubuntu/ bionic multiverse
deb http://archive.ubuntu.com/ubuntu/ bionic-updates multiverse
# deb-src http://archive.ubuntu.com/ubuntu/ bionic-updates multiverse

## N.B. software from this repository may not have been tested as
## extensively as that contained in the main release, although it includes
## newer versions of some applications which may provide useful features.
## Also, please note that software in backports WILL NOT receive any review
## or updates from the Ubuntu security team.
deb http://archive.ubuntu.com/ubuntu/ bionic-backports main restricted universe multiverse
# deb-src http://archive.ubuntu.com/ubuntu/ bionic-backports main restricted universe multiverse

## Uncomment the following two lines to add software from Canonical's
## 'partner' repository.
## This software is not part of Ubuntu, but is offered by Canonical and the
## respective vendors as a service to Ubuntu users.
# deb http://archive.canonical.com/ubuntu bionic partner
# deb-src http://archive.canonical.com/ubuntu bionic partner

deb http://security.ubuntu.com/ubuntu/ bionic-security main restricted
# deb-src http://security.ubuntu.com/ubuntu/ bionic-security main restricted
deb http://security.ubuntu.com/ubuntu/ bionic-security universe
# deb-src http://security.ubuntu.com/ubuntu/ bionic-security universe
deb http://security.ubuntu.com/ubuntu/ bionic-security multiverse
# deb-src http://security.ubuntu.com/ubuntu/ bionic-security multiverse
```
</details>

megabyte0, 2020-05-07:
You're right, and I think it's `ppa.launchpad.net/ondrej/php/ubuntu`, sorry I don't have access to the PC with all that packages installed anymore, so I can't tell...

closed 2020-07-16 by JacobHeAMD