GPUOpen-Drivers/AMDVLK #160: DOOM Eternal GPU hang on Cultist Base

wundrweapon, 2020-05-13:
Playing DOOM Eternal (via Proton 5.6-GE-2), the game consistently crashes the system in the level Cultist Base. Towards the end of the level, the player punches down a crucified demon with a large green mark on its chest. Half way through that animation, the crash occurs.

Having played this cutscene numerous times, it will always crash with AMDVLK and will never crash with RADV, without any other changes to my software setup. I switch drivers by uninstalling `mesa-vulkan-radeon` and installing `amdvlk` on Void Linux.

I'm willing to provide hardware and software information as necessary along with more tests, if someone can describe how they'd like them performed.

jinjianrong, 2020-05-13:
@wundrweapon thanks for reporting the issue. what GPU are you using? could you try to capture a renderdoc trace with RADV inthe frame that crashes with AMDVLK?

wundrweapon, 2020-05-15:
I had intended to gather some information for you today but Eternal pushed an update and it no longer runs in wine. All I can say is that I'm using a Vega 56, and that if anyone finds a way to run the game henceforth I will attempt the RADV RenderDoc trace. My apologies

alex9k1, 2020-05-24:
Seems like Denuvo Anti Cheat is getting removed: https://www.reddit.com/r/Doom/comments/gnjlo7/latest_information_on_update_1_anticheat/

wundrweapon, 2020-05-28:
The game is running again on Proton, but I seem to be a bit too dense to figure out how to get the RenderDoc trace you're looking for. If someone could point me to a quick guide to getting exactly what @jinjianrong is looking for, that would be much appreciated. I should also note that the crash doesn't just cause the game to exit; my whole system hangs. Hopefully that won't stop RenderDoc from doing its thing.

Ahmed-E-86, 2020-06-20:
To record renderdoc trace:

1- Install the native version of renderdoc.

2- Run the game like this from Steam:

ENABLE_VULKAN_RENDERDOC_CAPTURE=1 VK_ICD_FILENAMES=/etc/vulkan/icd.d/amd_icd64.json %command%

3- You need to press F12, or something like that to record renderdoc trace.

4- You will find the recorded file inside /tmp/RenderDoc

5- Upload it to your Google Drive, and copy the link of the file.

wundrweapon, 2020-07-19:
Those options cause the launch to fail. My initial guess is becase Vulkan is up to v1.2.141 whereas the JSON file (which for me is in `/usr/share/vulkan/icd.d/radeon_icd.x86_64.json`) points to API v1.2.131? But I'm not sure, and I don't know how to test what's *really* the cause. Any suggestions would be appreciated.  
Apologies for the delay.

jinjianrong, 2020-11-05:
@wundrweapon please help verify the fix with 2020.Q4.2 release

closed 2020-11-05 by jinjianrong

wundrweapon, 2020-11-09:
Negative, the hang persists on 2020.Q4.3  
When I have more free time I will re-attempt to trace on RenderDoc w/ Mesa RADV.

JaxLinAMD, 2020-11-09:
Hi @wundrweapon , do you download v-2020.Q4.3 package from Q4.3 release links? Or use "apt install amdvlk"? please notice that using apt install amdvlk, you will get Q4.1 installed.

wundrweapon, 2020-11-09:
Since I'm on Void I used `xbps-install`. I checked [the template file](https://github.com/void-linux/void-packages/blob/9151c507ce38b7a01eb7f047ea956298b02a4c4e/srcpkgs/amdvlk/template) and it does use Q4.3 (see lines 3 and 27 for reference).