GPUOpen-Drivers/AMDVLK #162: AMDVLK sets timezone to GMT+0 when vulkan loader just enumerates ICDs.

LizziAS, 2020-05-23:
issued discovered by Jakun at mangohud as reported here:  https://github.com/flightlessmango/MangoHud/issues/175

The basics are that i had a problem with my in game time in ESO switching to UTC, or gmt +0 which affected my in game clocks for ESO.  Was advised to try mangohud but then mangohud was showing one time for opengl and another for vulkan when using goverlay.  I'm not sure i understand why a graphics driver is setting the time to anything but i am sure there is some reason for it or it wouldnt be there.  However it is affecting my ability to game on linux as i need an in game clock that represents my local time.  Please advise..

A fuller explanation can be found in the link provided above as i initially submitted to mangohud for their help and Jackun has a far better understanding of what is specifically happening than i do...

Flakebi, 2020-05-25:
Interesting, the code that sets `TZ` is here: https://github.com/GPUOpen-Drivers/pal/blob/85170384d0ee5447ac267d478f7a21bc036ea71c/src/util/lnx/lnxArchiveFile.cpp#L87
I’ll fix this.

As a workaround, adding `AllowVkPipelineCachingToDisk,0` to `.config/amdPalSettings.cfg` should fix this too, at least vkcube does not set `TZ` anymore then.

LizziAS, 2020-05-25:
awesomesauce!  ty!

Flakebi, 2020-05-26:
A fix has been merged internally, so it should be in the release in 2–3 weeks.

LizziAS, 2020-05-27:
Sweet!  TY!

closed 2020-05-27 by LizziAS