GPUOpen-Drivers/AMDVLK #163: [DXVK] Deep Rock Galactic — compiling shader errors

RarogCmex, 2020-05-25:
See:
https://github.com/ValveSoftware/Proton/issues/284#issuecomment-633600400
```
[2020.05.25-14.20.45:650][  0]LogStandaloneRenderer: SlateD3DShaders::CompileShader() - D3DCompilerFunc Result: DXGI_ERROR_88760B59 [88760B59]
[2020.05.25-14.20.45:652][  0]LogStandaloneRenderer: GammaCorrectionCommon.hlsl:6:11: error: syntax error, unexpected NEW_IDENTIFIER
 Result: DXGI_ERROR_88760B59 [88760B59]
[2020.05.25-14.20.45:655][  0]LogStandaloneRenderer: SlateD3DShaders::CompileShader() - D3DCompilerFunc Result: DXGI_ERROR_88760B59 [88760B59]
[2020.05.25-14.20.45:657][  0]LogStandaloneRenderer: GammaCorrectionCommon.hlsl:6:11: error: syntax error, unexpected NEW_IDENTIFIER
 Result: DXGI_ERROR_88760B59 [88760B59]

```

```
23133.119:0120:0124:warn:seh:OutputDebugStringA "[2020.05.25-14.20.45:652][  0]LogStandaloneRenderer: GammaCorrectionCommon.hlsl:6:11: error: syntax error, unexpected NEW_IDENTIFIER\n Result: DXGI_ERROR_88760B59 [88760B59]\r\n"
23133.119:0120:0124:trace:seh:raise_exception code=40010006 flags=0 addr=0x7b00fc3e ip=7b00fc3e tid=0124
23133.119:0120:0124:trace:seh:raise_exception  info[0]=00000000000000af
23133.119:0120:0124:trace:seh:raise_exception  info[1]=00000000009bf850
23133.119:0120:0124:trace:seh:raise_exception  rax=00000000005fbfa0 rbx=00000000009bf850 rcx=00000000005fbf80 rdx=0000000000000000

```
After switching to RADV game runs

jinjianrong, 2020-05-26:
@RarogCmex could you help dump the shader that causes the compile error?

RarogCmex, 2020-05-27:
@jinjianrong I can't do it that moment

JaxLinAMD, 2020-08-28:
@RarogCmex  what version of OS are you using?
I didn't observe any issue with Ubuntu 18.04 v-2020.Q3.4,

jinjianrong, 2021-03-10:
Please re-open the issue if you still see the problem

closed 2021-03-10 by jinjianrong