GPUOpen-Drivers/AMDVLK #166: Multiple issues with Doom Eternal

Ahmed-E-86, 2020-06-27:
AMDVLK offers twice the performance I can get from RADV, but I encounter multiple issues:

1- GPU sometimes hangs during loading, and when trying to force-restart my system, I get these errors:

[image]

2- GPU hangs after I break a cracked wall like the wall in this picture:

[image]

3- I encounter an issue with the lighting when I am at the main menu:

[image]

jinjianrong, 2020-09-16:
@Ahmed-E-86 are you still seeing the issue with latest driver build?

Ahmed-E-86, 2020-09-17:
> @Ahmed-E-86 are you still seeing the issue with latest driver build?

Yes.

jinjianrong, 2021-03-09:
The game runs well now in our internal testing. Please help create a new issue if you still see issues

closed 2021-03-09 by jinjianrong